WiringPi/WiringPi #28: unregister ISR

DEvil0000, 2015-01-28:
wiringPiISR() will register a ISR. works well.

but then your program finishes execution. the program is not running anymore (ps -ef). you start a different program what is not using ISR at all.
it looks like the other program is kind of pausing execution at some point and waiting for a interrupt to occur. when you send signals on the pin used by the ISR the second program resumes.

if i set the ISR pin mode to output at the end of program one i will get process two running normal afterwards.

pinMode(Pin, OUTPUT);

lukaszjakubek, 2015-11-17:
Also noticed that GPIO pins are not unexported after program exits. I thing some cleanup function, which cancel interrupt treads and reset pins, will be appreciated.

sun-alf, 2015-11-17:
Please redirect your notices to paroga/WiringPi [forked from WiringPi/WiringPi]. "WiringPi/WiringPi" depot is not supported.

closed 2016-02-28 by Gadgetoid