WiringPi/WiringPi #30: wiringPiSPIDataRW  returns EINVAL

oklitz, 2015-01-28:
Hello !

I'm using arch linux: 3.18.3-3-ARCH #1 PREEMPT with newest wiringpi-git packet from the AUR.
module spi_bcm2708 is loaded and devices are in /dev/spidev0.0 and /dev/spidev0.1

	fd = wiringPiSPISetup (0, 500000);
	if (fd == -1)
		errExit("wiringPiSPISetup");
--works without errors

	uint8_t buff = 0b11110000;
	ret = wiringPiSPIDataRW (0, &buff, 1);
	if (ret == -1)
		errExit("wiringPiSPIDataRW");

returns -1 with [EINVAL Invalid argument]

echo 1 > /dev/spidev0.0  works fine

This code worked with older kernels, but after the jump to 3.18 it doesn't.
But i can't say it exactly because i have used the libwiringPi the last time in last year.

mfg marco

chrillomat, 2015-01-29:
Hi marco,
Can you please check if #31 solves your issues?
BR, Christoph

oklitz, 2015-03-15:
yes it does !
thanx

closed 2015-03-15 by oklitz