WiringPi/WiringPi #35: Unable to export GPIO17 (pin 11) on Raspberry Pi 2

aleksandaratanasov, 2015-08-25:
When I try to do `gpio export 17 out` I get

> gpio: Unable to open GPIO direction interface for pin 17: No such file or directory

I tried GPIO18 and GPIO22 and there is no problem. I know for sure that 17 is functional since I can access it using the RPi.GPIO lib and my LED is blinking accordingly.

I'm using [this tutorial](http://www.sirmc.net/view/9001).

sun-alf, 2015-08-25:
Please use this rep instead of root-rep:
https://github.com/paroga/WiringPi

From readMe:
Please note that the official way to get wiringPi is via git from
git.drogon.net and not GitHub.

ie.

  git clone git://git.drogon.net/wiringPi
http://wiringpi.com/

aleksandaratanasov, 2015-08-25:
I did try that out too but there is no Python version in there from what I can see. Or has the support for Python been dropped?

sun-alf, 2015-08-25:
I have no idea about that. But the only truth is that you don't have to use WiringPi in Python since it has a perfect library for GPIO: (now it is built-in and comes with Raspbian)
import RPi.GPIO as GPIO

aleksandaratanasov, 2015-08-25:
Ah, I thought that the RPi.GPIO was something different (at least from what I've read on the Internet). Btw the exporting of pins using the gpio application is important for C/C++ projects right? Because I do want to write some C/C++ code soon so I will be needing a decent GPIO abstraction layer so that I don't have to do it myself. :lollipop:

sun-alf, 2015-08-26:
If you want an app in c/c++, use the WiringPi then. =) It is the best lib I've found on the Internet. You don't have to make any GPIO exports and you don't even have to run your executable via 'sudo'.
But remember, official version of the lib is here:
http://wiringpi.com/

For Python experiments I use RPi.GPIO. There is no need to do any exports either.

aleksandaratanasov, 2015-08-26:
>you don't even have to run your executable via 'sudo'

When not running the compiled binary of my application with sudo, it doesn't work. WiringPi alone gives you an error message and tells you to run it with sudo.

As for the issue and exporting stuff:

> **wiringPiSetupSys (void);**
> This initialises wiringPi but uses the /sys/class/gpio interface rather than accessing the hardware directly. This can be called as a non-root user provided the GPIO pins have been exported before-hand using the gpio program

I'm currently using **wiringPiSetupGpio()** so it's not a problem but I was looking at all the options available and the one I've quoted requires the export step. The strange thing is that using **wiringPiSetupGpio()** and **wiringPiSetup()** work fine with GPIO17 (or pin 0 using the second (native to WiringPi) numbering).

aleksandaratanasov, 2015-08-27:
I got confused by all the various setups that wiringPi2 offers. The numbers `gpio` takes for pins are actually not the BCM but the standard pin numbers hence when I tried to export 17 I actually tried to export not GPIO17 but the 3.3v power pin, which corresponds to 17. I found this out when I tried exporting GPIO19 and got the same error when I entered 19. 19 on the board (non-BCM numbering) is the SPI_MOSI. I then checked my theory by exporting 26 and there were no errors. However when I tried to write a 1 (high) to it, my LED didn't light up. Then I changed 26 to 37 (corresponding to GPIO26), wrote a 1 and voila! The LED lit up.

I also found that the `gpio -g` allows me to use the BCM numbering. Neat!

aleksandaratanasov, 2015-08-27:
Nope. That's not it. The documentation states:

>This exports the given pin (BCM-GPIO pin number) as an input or output and makes it available for a user program running as the same user to use.

It seems there is some major inconsistency with the BCM numbering for this tool. I just tried to export 35 (GPIO19) and go an error.

sun-alf, 2015-08-27:
Oh, man... So much text! =)
WiringPi uses it's own pin numbering method. Look at this picture for Raspberry Pi A+:
http://pi4j.com/images/j8header-a-plus.png
(Corresponding WiringPi GPIO numbers are in bold)
You probably have to find something like this for your RPi2.

aleksandaratanasov, 2015-08-27:
Man, I write text because every time something screws up. I'm following the documentation on the website of WiringPi. There is no difference between RPi B+ and RPi 2 B in terms of pin layout whatsoever.

Also: http://wiringpi.com/wiringpi-updated-for-the-new-pi-v2/ (this update is from the 2nd of february this year)

closed 2015-08-27 by aleksandaratanasov

sun-alf, 2015-08-28:
Congratulations!