WiringPi/WiringPi #43: THIS IS NOT THE UP-TO-DATE REPOSITORY

ztane, 2016-02-21:
According [to the official page](http://wiringpi.com/download-and-install/), the repository is at git://git.drogon.net/wiringPi

savageautomate, 2016-02-21:
ztane is right, Gordon's repo, the official WiringPi repo is at: git://git.drogon.net/wiringPi

I have an up to date clone of the official repo here:
https://github.com/Pi4J/wiringPi

Thanks, Robert

Materdaddy, 2016-02-22:
That repository is giving a connection reset error message.  Any reason it is used instead of github?

ztane, 2016-02-22:
Because that is the author's repository - up to date with RPI2 and such, unlike this one which is an outdated fork (last update in 2013).

Materdaddy, 2016-02-23:
Is there something that can be cloned that is up to date?  I still cannot clone from his repository.

>  $ git clone git://git.drogon.net/wiringPi
> Cloning into 'wiringPi'...
> fatal: read error: Connection reset by peer

This is from multiple hosts.  His gitweb page works though: https://git.drogon.net/?p=wiringPi;a=summary

I work on a project that builds it from source, but I'm leaning toward just using a snapshot since his git server appears unreliable.

ztane, 2016-02-23:
@Materdaddy I concur, the git server seems to be down, however *this* snapshot is very broken on rpi2. It does not work at all but it also fails to tell that it does not work.

savageautomate, 2016-02-24:
My GitHub fork of WiringPi is current at:  https://github.com/Pi4J/wiringPi

I use this fork to build Pi4J (http://pi4j.com/) which is based on WiringPi.  I try to keep it up to date as much as possible.  No WiringPi changes (commits or pull-requests) are accepted to my fork, all updates flow directly from Gordon's source repo for WiringPi.

Thanks, Robert

Gadgetoid, 2016-02-27:
Now synced this from git.drogon.org for posterity. I haven't had the time to keep on the ball with WiringPi ports, but it's on my radar.

closed 2016-02-28 by Gadgetoid