WiringPi/WiringPi #80: The future of this wiringPi fork

Gadgetoid, 2020-09-18:
I'm posting this an issue to state the goals for the wiringPi org with the hope that those qualified to help and those more invested in the wiringPi ecosystem might be able to help steer this project toward them.

It's important to me, despite my lack of personal investment in this project, that:

1. This fork has some kind of future
2. It's clearly communicated that this is an unofficial fork, but also the canonical public version of the sourcecode
3. It's clearly communicated (perhaps in sweeping changes to the source headers) that this repository and the accompanying Discord are the canonical places to get help/support (*not* Gordon)

With respect to how this repository is maintained and organised, let's keep things simple and in order of priority:

1. Tag releases to give ports a stable version to submodule/link against/etc
2. Reduce busywork by adding continuous-integration and auto builds via GitHub actions
3. Keep on top of bugfix PRs, new Pi versions, etc
4. *Consider* new features and improvements

If you have a vested interest in keeping this wiringPi fork alive, and would like to join the org as a maintainer- let us know!

seamusdemora, 2020-11-19:
I'd like to see this wiringPi fork survive and thrive going forward. I have read Gordon's deprecation message - that's sad, but understandable. I wound up here because the `git` sources listed on Gordon's website don't appear to be functional any longer. I just have one question for now: what has happened to the [GPIO Utility](http://wiringpi.com/the-gpio-utility/)? Is it included in the sources here?

merwok, 2020-11-19:
It is here: https://github.com/WiringPi/WiringPi/tree/master/gpio
The `build` command at the root of the repo runs make in all subdirectories, including gpio.

seamusdemora, 2020-11-20:
@merwok:  Thanks - I didn't spend enough time looking. 

Another question: Apparently ver 5.5 of the Linux kernel will introduce new support for GPIO operations - it all sounds pretty nice (I've not tried it yet). Has the project team here made any decisions regarding the role these new 5.5 kernel features will play with WiringPi going forward? I've read at least one opinion that `wiringPi` will cease to be useful once 5.5 becomes "mainstream", but it seems to me there are always opportunities for software that makes hardware easier to use.

seamusdemora, 2021-08-08:
A [recent test posted on S.E.](https://raspberrypi.stackexchange.com/a/130530/83790) may add some impetus to maintaining and improving WiringPi. Briefly, the test results suggest that (by at least one measure) WiringPi's performance is an order of magnitude greater than the 2nd place library. Summarized briefly: 

```
WiringPi       7930214 toggles per second
RPi.GPIO        801556 toggles per second
lgpio           523049 toggles per second
pigpiod C         7014 toggles per second
pigpiod Python    5974 toggles per second
```

Gadgetoid, 2021-08-09:
Raw performance is useful in some cases, but doesn't exactly paint a fair picture with respect to RPi.GPIO which has to pay the cost of Python -> C on every single GPIO write. I'd venture that RPi.GPIO could attain speeds much closer to WiringPi if it implemented functions that kept the toggling entirely within the C domain.

Correct, cross-platform GPIO code should be written using libgpio and despite its performance WiringPi is largely an anachronism. If someone needs to toggle a pin at nearly 8MHz they're probably doing something enormously specific that shouldn't involve WiringPi at all.

WiringPi's USP was that it brought the Arduino-style of programming over to the Pi so that those users felt at home, and it does some pretty cool things with external devices mapped to virtual IO too. It clearly still has fans, users and software depending upon it and that's reason enough to keep it going.

seamusdemora, 2021-08-11:
When you need *"raw performance"* it doesn't matter if it's *fair* or not.

seamusdemora, 2021-08-12:
@merwok - 

Please refer to [this thread](https://github.com/RPi-Distro/repo/issues/214#issuecomment-897177001) on the RPi GitHub site.

merwok, 2021-08-12:
I don’t know why you’re tagging me, I’m not a maintainer here.

seamusdemora, 2021-08-13:
@merwok - my apologies, it was an error on my part.

joan2937, 2021-09-20:
> A [recent test posted on S.E.](https://raspberrypi.stackexchange.com/a/130530/83790) may add some impetus to maintaining and improving WiringPi. Briefly, the test results suggest that (by at least [...]

lgpio uses the new generic Linux /dev/gpiochip interface (libgpio to you).

pigpiod uses a client server model and each toggle requires a couple of network messages which accounts for the dramatic performance difference.  It would be more sensible to compare against pigpio.

seamusdemora, 2022-01-23:
@Gadgetoid : Would love to see someone do that... I am so *not-savvy* on what makes open-source projects viable, but it seems to me that if there is no *apt-installable package* of WiringPi, it will, like the old soldier, "just fade away".

Gadgetoid, 2022-02-09:
More because I need them than for any other reason, I have tagged a release and uploaded 2.61 debs for arm64 (Raspberry Pi OS 64bit) and armhf (Raspberry Pi OS). Would appreciate anyone with the time to battle test these!

seamusdemora, 2022-02-17:
Where to find the debs & how do I install one - the RPi OS one. I've a fairly recently upgraded RPi 3B+ to use for testing.

FWIW, I'm certain my testing won't be thorough, but I have a few scripts that I can run.

Gadgetoid, 2022-02-17:
@seamusdemora sorry I'm too GitHub native and forget that the files get attached waaaaaaaaaaaaaaaay down underneath all the release notes.

1. Go here: https://github.com/WiringPi/WiringPi/releases/tag/2.61-1
2. Scroooooooooooooooollll doooooooooooownnn

Alternatively since you're probably 32bit `armhf`you can grab it from this link: [wiringpi-2.61-1-armhf.deb](https://github.com/WiringPi/WiringPi/releases/download/2.61-1/wiringpi-2.61-1-armhf.deb)

Install with:

`sudo dpkg -i wiringpi-2.61-1-armhf.deb`

seamusdemora, 2022-03-08:
@Gadgetoid 

The 2nd link - for the 32-bit armhf gets me only a 404. Also tried w/ curl on RPi with no luck??

liffiton, 2022-03-08:
Ah, sorry -- I deleted the armhf .deb file because it wasn't actually a 32-bit armhf .deb file (reported in #159).  It was just the same file as the arm64 file.

I just rebuilt the armhf deb on a Pi myself and uploaded it to the release page.  Let me know if you have any issues with it.

seamusdemora, 2022-03-08:
@Gadgetoid, @liffiton,

Got it - thanks! FWIW, GitHub seems to obfuscate downloads, but this worked for me: 
```bash
curl -LO https://github.com/WiringPi/WiringPi/releases/download/2.61-1/wiringpi-2.61-1-armhf.deb
```

After my recent experience with the `libgpiod` (via user tools in `gpiod`), I am happy to have `wiringPi` (esp. `gpio`) restored on my Pi 3B+/bullseye. Hoping it gets restored to *package status* aga@Gadgetoid, @liffiton

seamusdemora, 2022-03-10:
@Gadgetoid @liffiton : 

Just an observation: After installing `wiringpi` with `dpkg` as above, I thought I'd check to see if wiringpi has made it into the `apt` repo for bullseye. Here's what I saw: 
```bash
pi@raspberrypi3b:~ $ apt-cache show pkg wiringpi
Package: wiringpi
Status: install ok installed
Priority: optional
Section: libraries
Maintainer: Gordon Henderson <projects@drogon.net>
Architecture: armhf
Version: 2.61
Depends: libc6
Description: The wiringPi libraries, headers and gpio command
  Libraries to allow GPIO access on a Raspberry Pi from C and C++
  and BASIC programs as well as from the command-line
Description-md5: 083689a41804acb3a354e723be4b8b2d
``` 

Was surprised to see that Gordon is still listed as the Maintainer!

Gadgetoid, 2022-03-10:
@seamusdemora that's weird. The `control` file is updated here, as is the script which generates it - https://github.com/WiringPi/WiringPi/blob/master/debian-template/wiringPi/DEBIAN/control

Could be something cached in apt, or just an older version getting packaged?

liffiton, 2022-03-10:
I think you just made the changes removing Gordon for 2.70, but the .debs were built for 2.61.

seamusdemora, 2022-03-13:
Another comment/question: Should there be a link on [the repo's "front page"](https://github.com/WiringPi/WiringPi) that takes one to the [Releases page](https://github.com/WiringPi/WiringPi/releases)?? 

Thanks,
~S

liffiton, 2022-03-16:
There is one.  It's part of Github's regular interface, and it's on the right hand side of the page (or below the README on mobile).

seamusdemora, 2022-03-17:
Ach!  Sorry - you're correct - I missed it.

seamusdemora, 2022-04-26:
This is my *go-to thread* for all WiringPi matters  :)  Apologies for being somewhat off-topic with this one, but...

I'm working on a bugfix to another RPi source package (my first time to try this). I've been using what I guess are the standard tools - specifically `debuild` to create the .deb file. However - I've just heard of what *sounds like* a "git-centric" tool: [git/gbp-buildpackage](https://manpages.debian.org/bullseye/git-buildpackage/gbp-buildpackage.1.en.html). What tool is the wiringpi team using?

mbhangui, 2023-06-21:
Wiring Pi is a noteworthy legacy left by Gordon. I have [forked](https://github.com/mbhangui/wiringPi) this project and made few changes. This fork is the best thing that has happened after development of the "official" version has ended. To summarize, I have made the following changes

1. Use GNU autotools to generate the libraries and the gpio utility. You just need to run ./configure;make && sudo make install
2. Incorporated few pull requests here into my fork. See [commit]([commit](https://github.com/WiringPi/WiringPi/commit/7b92b3bad9cc2f122be36cc4d207c04f86dd9bf5)
3. Added a repository on [Open Build Service](https://build.opensuse.org/package/show/home:mbhangui:raspi/wiringpi) for debian/RPM packages for Raspbian, debian and ubuntu distributions.
4. Created two scripts create_rpm and create_debian to generate RPM and debian files locally.
5. Created an action script to build the libwiringpi and libwiringpidev on commits to test the compilation.

I'm not sure of the future of this wiringPi fork. But I will be happy to create a pull request to submit my changes upstream to this project and I hope this project continues the legacy that Gordon has left. I'm reasonably good with C programming, GNU autotools and building RPM/Debian/Archlinux packages, but with zero knowledge of hardware side of things that Wiring Pi code is. Will be happy to contribute in any way I can.

sndnvaps, 2023-06-22:
@mbhangui  It's better to use Makefile or CMakeLists.txt to build it ... make it easy to cross compile.
just like Windows , it don't have bash shell and other cmd app, just make it easy to ,keep it simple stupid....

Gadgetoid, 2023-10-30:
@mbhangui I would be open to adding you as a maintainer here, but without a suitable counterpart who understands the hardware and GPIO side of this project I fear it would be pointless.

As discussed in https://github.com/WiringPi/WiringPi/issues/186 WiringPi needs a major rewrite around the Linux GPIO character device ABI - or probably libgpiod - if it's going to survive the transition to RPi OS Bookworm and Pi 5.

Without someone to make this transition, this project is effectively at the end of the dirt path beyond the end of the road.

I'm of the opinion that the entire WiringPi org should be archived, and we should focus our time on migrating the worthwhile concepts from WiringPi into newer, better libraries that are more thoroughly understood by their authors.

Gordon's legacy is some good ideas saddled with a bunch of baggage that nobody really has the time or inclination to understand. It's a waste of good talent trying to maintain it any further.

For my part- I have been working with libgpiod's author to ready the Python package for more general use.

As maintainers of WiringPi and its various ports, @liffiton @eugeneware @lrvick @nekuz0r @NoHomey @soarez what are your thoughts on this?

And @rafael2k @Lucino772 @GeVanCo @pixout - as contributors who have raised recent PRs in earnest here and might have a vested interest, do you have any input?

And more generally, is there anyone out there who could/would see WiringPi through the transition to the Linux GPIO character device ABI? (probably via `libgpio`)

pixout, 2023-10-30:
Currently fully loaded and no additional time for migration.

liffiton, 2023-10-31:
> As maintainers of WiringPi and its various ports, @liffiton [etc.] what are your thoughts on this?

I agree with everything you've said, basically.  Without anyone willing and able to perform the rewrite, we should archive all of this.  We can update READMEs and other points of contact (e.g., PyPI for the Python package) to be very clear on the deprecation status, what the alternatives are, and what OSes and hardware will still work with wiringpi.

Rewriting wiringpi feels like basically making a new library but keeping it under the same name.  That has some benefit in keeping old tutorials working, but that's not critical.  Anyone with a project that uses wiringpi can keep using older hardware and older kernels/OSes.

Lucino772, 2023-10-31:
I agree with you @Gadgetoid. Personally, I don't have the time to contribute to this project.

Gadgetoid, 2023-10-31:
Since it's likely that *more* people will jam up my GitHub notifications with ceaseless moaning about libgpio and Linux GPIO in general I am pulling the proverbial trigger.

I invite the maintainers of the various bindings to do the same, as and when they deem fit. Ping me if you have any questions or concerns.

If someone out there has a fork that they want the unruly masses of WiringPi users to be directed to, let me know and I will add some links into the README. Or you can just have ownership of this org :laughing:

rafael2k, 2024-02-29:
That is cool!
: )

mwallner, 2024-02-29:
> I'm maintaining my fork, as usual: https://github.com/Rhizomatica/WiringPi

that's cool! / we've been doing the same in the past but realized _this_ is where most people end up looking and we've had the chance to continue our work here :smiley: ,- you'd be more than welcome to participate here if you want to!
:wave:

pixout, 2024-02-29:
Awesome

rafael2k, 2024-02-29:
Sure.
: )

seamusdemora, 2024-03-10:
I just saw the [notice re ver 3.1 release](https://github.com/WiringPi/WiringPi/releases/tag/3.1).  My apologies, but I stopped following this thread after @Gadgetoid ***archived*** the repo last year, and so - I'm a little confused...  

A few questions to help me get my bearings: 
1. packaged as a `.deb`... does this mean that (once again) wiringPi can be installed using `deb`? 
2. what is the latest version it can be installed on? (bullseye, or... *bookworm*?!?) 
3. are the changes being made using the `libgpiod` library - or has it been cast aside??

I'd say this is pretty exciting news !

mwallner, 2024-03-11:
Hi @seamusdemora,
you're right, [wiringPi is back](https://github.com/WiringPi/WiringPi/issues/80#issuecomment-1970961372), we're doing our best to support all platforms as well as new OS versions (such as debian bookworm) - [see release 3.0](https://github.com/WiringPi/WiringPi/releases/tag/3.0)
currently we're not looking toward libgpiod because of it's devastating performance compared to wiringPi 😉 (DMA as long as it's feasible)
cheers ~ 🍸

rafael2k, 2024-03-11:
I migrated to gpiolib and performance is great... as fast as it can be:
https://github.com/raspberrypi/utils/tree/master/pinctrl

rafael2k, 2024-03-12:
I don't get when you say "Not quite". It is indeed direct access to hardware. I read the code.

I still use WiringPi for legacy projects.

mstroh76, 2024-03-12:
We are currently engaged in the process of developing additional documentation and conducting performance comparisons. However, it's important to note that our resources and time are limited, preventing us from parallelizing every aspect of our work.

rafael2k, 2024-03-12:
@ mwallner, if you could please read my message it would be nice. Yes, it can be measured, but none of the data you posted compares to gpiolib. Libgpiod has nothing to do with gpiolib, which is available at:
https://github.com/raspberrypi/utils/tree/master/pinctrl

If you need some high-speed switching of GPIOs, GPIOLIB is the way to go, not WiringPi. I'm sorry to correct you.

rafael2k, 2024-03-12:
Btw, I really appreciate you work, I really like WiringPi and I think it is easier to use and it is really wonderful it is back (and I plan to contribute too, as I mentioned, as I was maintaining my own fork).

I recently ported a software from WiringPi to gpiolib (for the reference: https://github.com/Rhizomatica/sbitx-core/tree/gpiolib ), and I can not stand by the fact the WiringPi can be faster than Raspberry Pi's reference implementation gpiolib.

mstroh76, 2024-03-12:
> @ mwallner, if you could please read my message it would be nice. Yes, it can be measured, but none of the data you posted compares to gpiolib. Libgpiod has nothing to do with gpiolib, which is [...]

Thanks for the input, good to have a comparison to WiringPi implementation. I already did all the RP1 implementation by my self.

rafael2k, 2024-03-12:
Cool. I'll do a PR to your Benchmark repo with gpiolib code for testing.
Good stuff, thanks for the references.

rafael2k, 2024-03-12:
So, your implementation is indeed much faster! I did a quick and dirty port of the Benchmark to gpiolib:
https://github.com/rafael2k/Benchmark

While in Pi 4 the difference is not much, something is very slow in the Pi 5 gpiolib implementation. Now I got puzzled why. Will investigate further. Anyway... I'll revert to using WiringPi. Thanks a lot @mwallner.

mwallner, 2024-03-12:
I've realized that this issue will soon become too cluttered to be of any use, so I'm going to lock it down.

* I've re-enabled [discussions](https://github.com/WiringPi/WiringPi/discussions), so feel free to engage there!
* You can always create new [issues](https://github.com/WiringPi/WiringPi/issues).
* Contributions are welcome!

TLDR:
[WiringPi is not dead.](https://github.com/WiringPi/WiringPi/issues/80#issuecomment-1970961372) With [version 3](https://github.com/WiringPi/WiringPi/releases), a new group of maintainers took over, aiming to keep the arguably best GPIO library around for as long as possible.

closed 2024-04-12 by mwallner