WiringPi/WiringPi #82: 2.52 version for ubuntu on rpi4B

MrRivi, 2020-09-23:
Hi,

I don't know if this is the correct place to ask but I would like to know about the 2.52 version of WiringPi. I saw Gordon posted on his blog that there's a new update: http://wiringpi.com/wiringpi-updated-to-2-52-for-the-raspberry-pi-4b/

Is anybody working on the update for this mirror? 

I am trying to make the 2.52 update work with Ubuntu 16.04.6 running on a rpi4B (following the instructions on Gordon's blog) but it doesn't upgrade.

Any info is welcomed. Thank you in advance.

[image]

liffiton, 2020-09-23:
As far as I know, Gordon never released source code for that update.  However, the code in this repository has been updated (as a fork of Gordon's code) to support the latest RPi.

andrej-83, 2020-10-04:
Hey i am not sure if i Understand you right. If i am right you try to upgrade your wiringpi version to 2.52 on a Raspberry 4?
If yes try.

cd /tmp
wget https://project-downloads.drogon.net/wiringpi-latest.deb
sudo dpkg -i wiringpi-latest.deb

# check Version with

gpio -v

PPCM, 2020-12-07:
In my case, the installation on Ubuntu aarch64 is not possible too
```
ubuntu@ubuntu:/tmp$ wget https://project-downloads.drogon.net/wiringpi-latest.deb
--2020-12-07 02:50:11--  https://project-downloads.drogon.net/wiringpi-latest.deb
Resolving project-downloads.drogon.net (project-downloads.drogon.net)... 188.246.205.22, 2a03:9800:10:7b::2
Connecting to project-downloads.drogon.net (project-downloads.drogon.net)|188.246.205.22|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 52260 (51K) [application/x-debian-package]
Saving to: ‘wiringpi-latest.deb’

wiringpi-latest.deb             100%[=======================================================>]  51.04K  --.-KB/s    in 0.06s

2020-12-07 02:50:12 (812 KB/s) - ‘wiringpi-latest.deb’ saved [52260/52260]

ubuntu@ubuntu:/tmp$ sudo dpkg -i wiringpi-latest.deb
dpkg: error processing archive wiringpi-latest.deb (--install):
 package architecture (armhf) does not match system (arm64)
Errors were encountered while processing:
 wiringpi-latest.deb
```

My ubuntu version
```
ubuntu@ubuntu:/tmp$ uname -a
Linux cluster01 5.8.0-1007-raspi #10-Ubuntu SMP PREEMPT Thu Nov 5 17:52:40 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux
```

Any idea?

liffiton, 2021-02-19:
@PPCM, WiringPi is no longer maintained by the original author at drogon.net.  This repository is a fork, and I'm afraid it does not currently produce a .deb package.  You can download and build from source following the instructions in INSTALL.

The original issue is more of a question, and that has been answered, so I guess I'll go ahead and close this issue.

closed 2021-02-19 by liffiton