WiringPi/WiringPi #83: No longer builds with gcc10

gearhead, 2020-09-26:
I am running this on Arch Linux. It no longer builds as Arch has updated to gcc10 (currently at gcc 10.2.0) and it fails with this linker error at build.
```
[Link (Dynamic)]
/usr/bin/ld: wpiExtensions.o:(.bss+0x408): multiple definition of `comDat'; drcNet.o:(.bss+0x400): first defined here
collect2: error: ld returned 1 exit status
make: *** [Makefile:78: libwiringPi.so.2.60] Error 1
```
I tried a few things based on some internet googling, but cannot get it past this step.
FWIR the unaccepted hack is to add -fcommon to CFLAGS as I could not figure out the preferred way according to this page:
[Gentoo Wiki](https://wiki.gentoo.org/wiki/Gcc_10_porting_notes/fno_common)
If I go into the wiringPi directory and add '-fcommon' to the CFLAGS, it will build libwiringPi.so.2.60 but when I do the same in the wiringPiD directory, I get a libcrypt error.
```
[Link]
/usr/bin/ld: warning: libcrypt.so.1, needed by /usr/lib/gcc/aarch64-unknown-linux-gnu/10.2.0/../../../../lib/libwiringPi.so, may conflict with libcrypt.so.2
```

gearhead, 2020-09-26:
I finally figured it out. The attached patch will allow a build under gcc 10.2
[gcc_10_patch.zip](https://github.com/WiringPi/WiringPi/files/5287051/gcc_10_patch.zip)

liffiton, 2020-09-26:
Thanks for the report.

Pull request #76 corrects this, too.  It's pending review.  I'd prefer to merge a cleaned up version of that pull request, but if it sits too long, I might just make the fix separately.

liffiton, 2020-11-27:
Fixed in #76.

closed 2020-11-27 by liffiton