WiringPi/WiringPi #88: Pi 400 and CM4 support

fanoush, 2020-11-21:
This patch was needed to support Pi400 (tested) and CM4 (untested)
See also https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md
```
diff --git a/debian/rules b/debian/rules
old mode 100644
new mode 100755
diff --git a/wiringPi/wiringPi.c b/wiringPi/wiringPi.c
index c3b9d8b..59e1baf 100644
--- a/wiringPi/wiringPi.c
+++ b/wiringPi/wiringPi.c
@@ -219,7 +219,7 @@ volatile unsigned int *_wiringPiTimerIrqRaw ;
 
 static volatile unsigned int piGpioBase = 0 ;
 
-const char *piModelNames [20] =
+const char *piModelNames [21] =
 {
   "Model A",   //  0
   "Model B",   //  1
@@ -240,7 +240,8 @@ const char *piModelNames [20] =
   "CM3+",      // 16
   "Pi 4B",     // 17
   "Unknown18", // 18
-  "Unknown19", // 19
+  "Pi 400",    // 19
+  "CM4",       // 20
 } ;
 
 const char *piRevisionNames [16] =
@@ -2312,6 +2313,8 @@ int wiringPiSetup (void)
       break ;
 
     case PI_MODEL_4B:
+    case PI_MODEL_400:
+    case PI_MODEL_CM4:
       piGpioBase = GPIO_PERI_BASE_2711 ;
       piGpioPupOffset = GPPUPPDN0 ;
       break ;
diff --git a/wiringPi/wiringPi.h b/wiringPi/wiringPi.h
index ecce383..0800ad2 100644
--- a/wiringPi/wiringPi.h
+++ b/wiringPi/wiringPi.h
@@ -103,6 +103,8 @@
 #define        PI_MODEL_3AP            14
 #define        PI_MODEL_CM3P           16
 #define        PI_MODEL_4B             17
+#define        PI_MODEL_400            19
+#define        PI_MODEL_CM4            20
 
 #define        PI_VERSION_1            0
 #define        PI_VERSION_1_1          1
@@ -114,7 +116,7 @@
 #define        PI_MAKER_EMBEST         2
 #define        PI_MAKER_UNKNOWN        3
 
-extern const char *piModelNames    [20] ;
+extern const char *piModelNames    [21] ;
 extern const char *piRevisionNames [16] ;
 extern const char *piMakerNames    [16] ;
 extern const int   piMemorySize    [ 8] ;
```

TheFlav, 2020-11-25:
I added this patch so I could use 'gpio readall' on the CM4.  To do it, you will also need to edit gpio/readall.c and add (model == PI_MODEL_CM4) and (model == PI_MODEL_400) to the if statements in void doReadall (void)

fanoush, 2020-11-26:
Yes, indeed, missed that. I should have done search for it in all repo.  Looks like it was the last place to fix https://github.com/WiringPi/WiringPi/search?q=PI_MODEL_4B  I have fixed it locally, will make PR here with complete change.

closed 2020-11-26 by Gadgetoid

TheFlav, 2020-11-26:
I didn't research it much further, but, after setting all that, the output from 'gpio readall' seemed off by 1 GPIO.  'gpio allreadall' seemed correct.  The code for 'readall' seemed to just call allreadall, but then I switched to using 'raspi-gpio get' which worked well for what I was doing.  It seemed like gpio wasn't setting the pullups properly, but (again) I didn't research it much.

4neta, 2022-09-03:
> I added this patch so I could use 'gpio readall' on the CM4. To do it, you will also need to edit gpio/readall.c and add (model == PI_MODEL_CM4) and (model == PI_MODEL_400) to the if statements in void doReadall (void)

It still doesn't work for my Pi 400, but it's quite an old issue. Are there any other code changes necessary now?

PR #89 (2020-11-26): Add support for Pi 400 and CM4
- just use same GPIO_PERI_BASE_2711 as Pi4
- add to readall - 400 as 4B, CM4 as previous CM

Seems to work for me with Pi400, don't have CM4 to test.

resolves #88