WiringPi/WiringPi #93: Using Pi4J 1.2 Java Library with Raspberry 4 - Sharing Docker file to produce : libpi4j.so

fguiet, 2020-12-05:
Hi,

You want to use Pi4J 1.2 Java Library with Raspberry 4 and arm64 architecture?

You know that last version of Pi4J (1.2 as of 2020/12/05) contains the JNI library called `libpi4j.so` build for armhf architecture...So I would like to share this Dockerfile : <https://github.com/fguiet/automationserver-v2/blob/main/libpi4j/builder/Dockerfile>

Just run :

```bash
docker build -t libpi4j .
docker run -it --rm -v /tmp:/tmp libpi4j /bin/bash -c "cp -r /applications/builder/output/libpi4j.so /tmp"
```

You'll get the `libpi4j.so` in your /tmp folder...built again last version of WiringPi (Raspberry 4 ready).
All you have to do to use the Pi4J Java library is to replace `lib\raspberrypi\dynamic\libpi4j.so` in the Ri4J 1.2 Jar file that you can get from Maven Repository.

You can build the WiringPi on your distro by running : 

```bash
cd automationserver-v2/libpi4j/builder/vendor/WiringPi/master-20201205 && chmod u+x build && ./build
```

Now you can use the Pi4J 1.2 Jar file in your raspberry 4 java project to control GPIO :) 

Use it at your own risk :) 

I just tested on GPIO 2 (PJ4J numbering scheme) on a debian 10 distro with OpenJDK 11 installed ... so far so good ! 

My two cent,
Fred

bennymeg, 2021-03-09:
Since Jan 2021 you can use [Pi4J 1.3](https://pi4j.com/1.3/)

savageautomate, 2021-03-09:
Pi4J v1.4 is also released now if you need to use with newer JDK (11).

https://pi4j.com/1.4/

closed 2021-11-22 by liffiton