WiringPi/WiringPi #99: Is this compatible with other mini computers such as the Odroid?

Joe23232, 2021-02-15:
I know this is compatible with the Raspberry Pi, but is it also compatible with the Odroid and other mini computers?

Furthermore is this compatible with the Arduino? If not then how similar is this library compared to the Arduino?

liffiton, 2021-02-19:
It is definitely not compatible with Arduino.  The Arduino is a completely different type of system than a Raspberry Pi.  It's difficult to say how it compares, since there are so many differences, but the main one is that an Arduino will run nothing but the code you put on it, while a Raspberry Pi is a full computer running a full OS and many different programs at once.

As far as I know, this library will not be compatible with other RPi-like computers.  It's fairly specifically tied to the hardware and the CPU of the Raspberry Pis.

closed 2021-02-19 by liffiton

Joe23232, 2021-02-19:
I see thanks man