WiringPi/WiringPi #100: wiringpi reaps unrelated processes when registering interrupt handler

freddyrios, 2021-02-23:
The problem is specifically in [these lines](https://github.com/WiringPi/WiringPi/blob/e9821abdb4b4fe46a2ea9243471d339435fa7bde/wiringPi/wiringPi.c#L2033-L2053), where some child process is started but then unrelated processes are reaped with `wait(NULL)`

There is more context on the thread where someone from [microsoft answered such a call reaps other processes](https://github.com/dotnet/runtime/issues/33297#issuecomment-784269606). There is some good context on my comment before that + a nice trace using strace earlier.

liffiton, 2021-02-23:
Yup, that looks like a pretty clear bug.

Are you able to build and test code with a potential fix?  I think it should just be a one line change, but I'm unfortunately not set up to test that change myself.

freddyrios, 2021-02-23:
@liffiton I don't have access to the environment until tomorrow at the office. I was kind of ready to move on as we now run the register at a point where there are no other child processes, but if there is a fix I can try it. 

If I were to build from source, is it enough to clone the repo in a pi with nothing other than raspbian lite and run https://github.com/WiringPi/WiringPi/blob/master/build?

I would need to do an extra build without the change too, as the reproduction of the issue is so time sensitive that I am concerned that it could stop hitting the right timing to reproduce just due to differences on any tooling used to build it.

freddyrios, 2021-02-23:
@liffiton so at the end I was able to confirm today that I can build from source and reproduce the issue with the newly built library.

Do you have a proposed patch / branch?

freddyrios, 2021-02-23:
@liffiton once I had tried a local build I realized I could as well take the extra minutes to get the right call in place.

liffiton, 2021-02-24:
Thanks for doing that and confirming the fix!  I appreciate the help.

closed 2021-02-24 by liffiton