WiringPi/WiringPi #107: Archlinux ARM "Unable to determine board revision from /proc/cpuinfo"

DarK-rtfm, 2021-03-28:
I have archlinux arm installed on my raspberry pi 4, but I get the following error while trying to use wiringpi:
```
Oops: Unable to determine board revision from /proc/cpuinfo
 -> No "Hardware" line
 ->  You'd best google the error to find out why.
```
I tried both the python library and the [library for haskell](https://hackage.haskell.org/package/wiringPi), but both has resulted in the error above.
Here is my `/proc/cpuinfo`:
```
processor	: 0
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3
```

liffiton, 2021-03-30:
I'm going to guess that Arch configures its kernel differently than Raspbian, which results in the contents of `/proc/cpuinfo` differing in this way.

The hardware check in [`piGpioLayout()`](https://github.com/WiringPi/WiringPi/blob/10be9486ffbaf0337aefb3033e342bc24e3fc0f6/wiringPi/wiringPi.c#L747) needs to be updated to handle this format (and to clean it up in general), but I'm afraid that requires knowledge and testing resources I (as a barely-here contributor) don't have.  This issue might remain until a maintainer steps up who can handle it.

If you're comfortable editing and recompiling the code, a hacky workaround might be to modify `piGpioLayout()` to just return `2`, assuming you're not running this on a very early board.

Alternatively, if it's possible to run a Raspbian or related kernel on Arch, that might get you the contents of `/proc/cpuinfo` that the current code expects.

albertmonzo, 2021-04-11:
I'm getting the same error with Debian 10 on Raspberry Pi 4, WiringPi 2.60. Tried to hardcode `return 2` as suggested, but then it fails because of "No Revision Line". 

This isn't as easy a hack because the function sets a bunch of magic values that I'm not familiar with. I did enjoy reading through all the angry comments about the emails though...

liffiton, 2021-04-11:
@albertmonzo, I meant you could make the `piGpioLayout()` function return 2 immediately, in which case it shouldn't even get to the "No Revision Line" check.  And unless I'm missing something, the function does not modify any external state -- its only purpose is to return 1 or 2.  So hardcoding the function to return 2 *should* work (except for breaking it when run on early board versions, of course).

If you add `return 2` as the first line of the function and that doesn't work somehow, let me know.

Thanks for the added report that the existing code fails in Debian 10, too.  Just to be clear, that was plain Debian, and not Raspberry Pi OS (originally known as Raspbian), correct?

DarK-rtfm, 2021-04-11:
@liffiton `piGpioLayout()` only raises "No Hardware Line". After modifying it to return 2, another method, `piBoardId()` (Which calls `piGpioLayout()` as a check) is responsible for the "No Revision Line" error. I believe this is the method @albertmonzo meant, as this is the part where `model`, `rev`, `mem`, and  `maker`  are set.

I tried changing `piBoardId()` to set everything to zero, which is the last fallback at the end of the method, but this results in the following runtime error: `wiringPiSetup: mmap (GPIO) failed: Operation not permitted`. I am not sure what this means, but I assume that this is caused by WiringPi trying to access incorrect part of memory due to wrong values set. Maybe manually setting the correct values will work?

liffiton, 2021-04-11:
Ah yes I see.  Thanks for clarifying.  So there's more than just the check in `piGpioLayout`.  Then even any workaround will be more involved, and I'm afraid we're back to "This issue might remain until a maintainer steps up who can handle it."

You're right, just figuring out and setting the correct values in `piBoardId` looks like it would be a workaround.  The best way I can imagine to do that would be to run the code on the board under Raspberry Pi OS, which I'm guessing (I could be wrong) still reports the board revision the way this code expects.  With `wiringPiDebug` set to true, [this code](https://github.com/WiringPi/WiringPi/blob/f66c883d7c75280971a01619cd503d1809754801/wiringPi/wiringPi.c#L1031) would then print out the values you need, I think?

But really, unless someone is going to maintain and update this codebase, I'm sorry to say it's probably a better idea to find alternatives.  Maybe it will continue to be compatible with Raspberry Pi OS, but even that is not guaranteed going forward.

seamusdemora, 2021-04-11:
@liffiton : I think that's a perfectly good approach. WiringPi was (AFAIK) written for Raspbian, and given the state of things, taking on another OS sounds "ambitious".

DarK-rtfm, 2021-04-11:
I got it working.
I ended up asking for a copy of my friend's `/proc/cpuinfo`, and I simply made wiringpi use this fake one instead of my actual one. Not sure if my friend has the exact same model, or just a "good enough" match, but it works like a charm.
Also, I guess this kind of solution would work on any linux distro as long as the fake cpuinfo file is correct about the hardware.

albertmonzo, 2021-04-11:
It's strange because the codebase defines PI_MODEL_4B and has conditional logic based on this value but it's never set. Nevertheless, I hardcoded the return of `piBoardId()` to PI_MODEL_4B,0,0,0 seeing that the rest of the values don't seem to matter and finally got past the error:

```
 gpio -v
gpio version: 2.60
Copyright (c) 2012-2018 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Pi 4B, Revision: 00, Memory: 256MB, Maker: Sony
  * Device tree is enabled.
  *--> Raspberry Pi 4 Model B Rev 1.2
  * Root or sudo required for GPIO access.
```

But I still can't get it working on plain Debian 10, getting this:

```
wiringPiSetup: mmap (GPIO) failed: Operation not permitted
```

Tried to add `iomem=relaxed` to `/boot/firmware/cmdline.txt` as suggested in the [Debian Wiki](https://wiki.debian.org/RaspberryPi4#GPIO_support) and running as root but it didn't make any difference.

liffiton, 2021-04-11:
@albertmonzo, I've opened another issue for that, which is a separate problem.  They're linked, but I think it's cleaner to discuss each separately.

Gadgetoid, 2021-04-12:
Bit of a tangent, but we should definitely remove this line: https://github.com/WiringPi/WiringPi/blob/10be9486ffbaf0337aefb3033e342bc24e3fc0f6/wiringPi/wiringPi.c#L788  :laughing: 

Gordon's enjoying the country with his horse these days and probably has no inclination to hear a peep from users of our on-life-support zombie of his library :grimacing:

And... this is pretty dicey:

https://github.com/WiringPi/WiringPi/blob/10be9486ffbaf0337aefb3033e342bc24e3fc0f6/wiringPi/wiringPi.c#L1058-L1088

As far as sniffing the Raspberry Pi revision on Arch is concerned... I don't see an easy way to do it but IIRC this song and dance is used to get the right memory offsets for GPIO memory mapping. This information should be more directly available from `/proc/device-tree/soc/ranges` and `/proc/device-tree/axi/vc_mem/reg`. I wrote some functions to unpack this data for rpi_ws281x a while (nearly 6 years!?) ago and they might be salient for WiringPi - https://github.com/jgarff/rpi_ws281x/pull/104/files

closed 2023-10-31 by Gadgetoid

pasquale90, 2024-03-08:
I can;t be sure if this is adequate, but I managed to resolve this using [this workaround](https://github.com/WiringPi/WiringPi/issues/108#issuecomment-817384407). I also replaced the line
````
*model = PI_MODEL_4B ; *rev = PI_VERSION_1_2 ; *mem = 2 ; *maker = PI_MAKER_EGOMAN;
```
with
```
*model = PI_MODEL_3B ; *rev = PI_VERSION_1_2 ; *mem = 2 ; *maker = PI_MAKER_EGOMAN;
```
...., as I use a rpi 3B. After changing the file, I recompiled and the runtime error disappeared.

mstroh76, 2024-03-08:
Please make a new issue and attach the output of /proc/cpuinfo, and the OS you use. We are only testing with Raspberry Pi OS (Bookworm and partially Bullseye)

mstroh76, 2024-03-08:
Con you please execute https://github.com/GrazerComputerClub/Benchmark/blob/master/src/pidetect.py and post the output

mstroh76, 2024-03-08:
I created #194, please don't report to closed issues.