WiringPi/WiringPi #108: Debian 10: wiringPiSetup: mmap (GPIO) failed: Operation not permitted

liffiton, 2021-04-11:
[To workaround issue#107] I hardcoded the return of `piBoardId()` to PI_MODEL_4B,0,0,0 seeing that the rest of the values don't seem to matter and finally got past the error:

```
 gpio -v
gpio version: 2.60
Copyright (c) 2012-2018 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Pi 4B, Revision: 00, Memory: 256MB, Maker: Sony
  * Device tree is enabled.
  *--> Raspberry Pi 4 Model B Rev 1.2
  * Root or sudo required for GPIO access.
```

But I still can't get it working on plain Debian 10, getting this:

```
wiringPiSetup: mmap (GPIO) failed: Operation not permitted
```

Tried to add `iomem=relaxed` to `/boot/firmware/cmdline.txt` as suggested in the [Debian Wiki](https://wiki.debian.org/RaspberryPi4#GPIO_support) and running as root but it didn't make any difference.

_Originally posted by @albertmonzo in https://github.com/WiringPi/WiringPi/issues/107#issuecomment-817378545_

liffiton, 2021-04-11:
One possible cause may be that [this line setting piGpioBase to 0](https://github.com/WiringPi/WiringPi/blob/10be9486ffbaf0337aefb3033e342bc24e3fc0f6/wiringPi/wiringPi.c#L2338) was triggered.  It's not clear that the logic there is correct, because it then proceeds to use `piGpioBase=0` in the `mmap` calls, which doesn't seem correct.

albertmonzo, 2021-04-11:
That did it! Here's the patch if anyone else needs it:

```
diff --git a/wiringPi/wiringPi.c b/wiringPi/wiringPi.c
index d45f83f..6f3543d 100644
--- a/wiringPi/wiringPi.c
+++ b/wiringPi/wiringPi.c
@@ -750,6 +750,8 @@ int piGpioLayout (void)
   char line [120] ;
   char *c ;
   static int  gpioLayout = -1 ;
+
+  return 2;

   if (gpioLayout != -1)	// No point checking twice
     return gpioLayout ;
@@ -964,6 +966,9 @@ void piBoardId (int *model, int *rev, int *mem, int *maker, int *warranty)
 //	Will deal with the properly later on - for now, lets just get it going...
 //  unsigned int modelNum ;

+  *model = PI_MODEL_4B ; *rev = PI_VERSION_1_2 ; *mem = 2 ; *maker = PI_MAKER_EGOMAN;
+  return
+
   (void)piGpioLayout () ;	// Call this first to make sure all's OK. Don't care about the result.

   if ((cpuFd = fopen ("/proc/cpuinfo", "r")) == NULL)
@@ -2335,7 +2340,6 @@ int wiringPiSetup (void)
   {
     if ((fd = open ("/dev/gpiomem", O_RDWR | O_SYNC | O_CLOEXEC) ) >= 0)	// We're using gpiomem
     {
-      piGpioBase   = 0 ;
       usingGpioMem = TRUE ;
     }
     else
```

liffiton, 2021-04-12:
Great!  There may be other pitfalls lurking, so feel free to open other issues if any arise, of course.

And while it's possible that removing the `piGpioBase = 0` line there is the right call in general, I'm not confident enough to commit that myself.  So I'll leave this open to help others find a potential solution, at least.

liffiton, 2021-04-12:
As I read the code further, I see how `piGpioBase = 0` could make sense.  If that is set, the code will also be using the file descriptor `fd` that opened the `/dev/gpiomem` file when it calls `mmap` later.  It is reasonable that a base address of 0 within that file would be the right thing to do.  It depends on how `/dev/gpiomem` works...

The discussion and code linked [here](https://github.com/raspberrypi/linux/issues/3055) suggests the offset doesn't matter and is in fact ignored.  But then why did changing the offset make a difference for you?  My guess is that it may be another kernel difference between Debian and Raspberry Pi OS.

That discussion and the kernel code linked, combined with your result, suggests more strongly that leaving `piGpioBase` unchanged when using `/dev/gpiomem` is generally the right thing to do.  I'm still not certain, though.  :smile:

closed 2023-10-31 by Gadgetoid

Xrockyman, 2024-04-16:
> Tried to add `iomem=relaxed` to `/boot/firmware/cmdline.txt` as suggested in the [Debian Wiki](https://wiki.debian.org/RaspberryPi4#GPIO_support) and running as root but it didn't make any difference.
> 

In my case (Debian GNU/Linux 12 (bookworm)) I had the same problem and adding  `iomem=relaxed` to `/boot/firmware/cmdline.txt` solved the problem. Thank you.