WiringPi/WiringPi #118: 我在树莓派里面用build debian指令生成的deb安装不了

kuailedeluojie-sudo, 2021-07-13:
wiringpi:arm64 : Depends: libc6:arm64 but it is not installable

liffiton, 2021-07-15:
Google [translates](https://translate.google.com/?sl=zh-CN&tl=en&text=%E6%88%91%E5%9C%A8%E6%A0%91%E8%8E%93%E6%B4%BE%E9%87%8C%E9%9D%A2%E7%94%A8build%20debian%E6%8C%87%E4%BB%A4%E7%94%9F%E6%88%90%E7%9A%84deb%E5%AE%89%E8%A3%85%E4%B8%8D%E4%BA%86&op=translate) the title as "I can’t install the deb generated by the `build debian` command in the Raspberry Pi ."

[I'm afraid I don't have access to a Pi at the moment to test or reproduce this.  I just wanted to get the translating in here in case someone else can reproduce it or provide any ideas.]

liffiton, 2021-11-22:
With no further details, I'm going to close this.  Feel free to re-open this or a new issue with more information if the problem remains.

closed 2021-11-22 by liffiton

LumenYoung, 2023-07-13:
I don't intend to reopen this issue again, but I think I know what is the problem from this error message, I'm commenting for anyone who has encountered the same issue in the future:

This might be that you are trying to install a arm64 pkg on armhf system. I'm using raspberry pi 4B and installed a ubuntu system from rpi-image-writer. I assumed this ubuntu to be arm64, but in fact you can check your architecture by running `dpkg --print-architecture`, ensure you are using the pkg for appropriate architecture