WiringPi/WiringPi #119: Compute Module 4 support is wrong

toca21, 2021-07-20:
I did not get my project to work when migrated to CM4 from RpiZ. So I examined the CM4 support in WiringPi and found out that CM4 is treated as CM3 and that is not correct in terms of pin numbering afaik. CM4 is more like standard RPI4 etc.

So I removed the statement model == PI_MODEL_CM4 in below code inside WiringPi.c:

```
 if ((model == PI_MODEL_CM) ||
      (model == PI_MODEL_CM3) ||
      (model == PI_MODEL_CM3P) ||
      (model == PI_MODEL_CM4))
    wiringPiMode = WPI_MODE_GPIO ;
  else
    wiringPiMode = WPI_MODE_PINS ;
```

and also moved the CM4-define to correct place in readall.c as below:

```
if ((model == PI_MODEL_A) || (model == PI_MODEL_B))
    abReadall (model, rev) ;
  else if ((model == PI_MODEL_BP) || (model == PI_MODEL_AP) ||
	(model == PI_MODEL_2)    ||
	(model == PI_MODEL_3AP)  ||
	(model == PI_MODEL_3B)   || (model == PI_MODEL_3BP) ||
	(model == PI_MODEL_4B)   || (model == PI_MODEL_400) ||
	(model == PI_MODEL_ZERO) || (model == PI_MODEL_ZERO_W) || (model == PI_MODEL_CM4))
    piPlusReadall (model) ;
  else if ((model == PI_MODEL_CM) || (model == PI_MODEL_CM3) || (model == PI_MODEL_CM3P))
    allReadall () ;
  else
    printf ("Oops - unable to determine board type... model: %d\n", model) ;
```

I compiled and got it working all the way. I could do a pull req if you agree that this is correct?

aurelian17, 2021-07-20:
If you can submit this fix would be great. I had the same issue with my CM4 setup, but I didn't have the time to investigate and potentially fix the problem. Thanks in advance.

toca21, 2021-07-20:
I'll fix! Do you want the pull directly to master or wich branch should I use?

liffiton, 2021-07-20:
Hi, @toca21.  Thanks for the report and for offering a fix.  A pull request to master is okay; we don't have a dev branch or anything.  And can you point to any reference that shows the differences you're talking about?

@aurelian17, would you be able to test the fix?

I'm asking because I'm around here to maintain the repo a bit to help folks out, but I'm no longer using it, nor do I have an RPi4 to test with.

Thanks!

toca21, 2021-07-20:
Hi Mark
It's a small fix but as I mention in the PR I have only verified GPIO, Serial and i2c.

aurelian17, 2021-07-21:
@liffiton Yes, I have the hardware and will be able to test at least the GPIO, serial and SPI.

toca21, 2021-07-21:
Great, just let me know if I should help out in any way guys.

liffiton, 2021-07-26:
Thanks!

@aurelian17 , if you test the fix successfully yourself (from the pull request), I'll go ahead and merge it.  (The simplest way might just be to clone https://github.com/toca21/WiringPi and build/test from there.)  And if not, well, I'll probably just merge it anyway since it's simple enough.

aurelian17, 2021-07-26:
Hi Mark,

I'll do the testing next weekend. I'm a little bit busy with a release
this week. You can merge it and I can trigger a new build on my side and
come back to you with my feedback.

Thanks,
Aurelian

liffiton, 2021-07-26:
Sure, that works for me.  I'll merge the pull request but leave this issue open to hear back from you.  Thanks.

aurelian17, 2021-08-04:
Sorry for the late reply. I finally made my CM4 setup working and I tested
today on a yocto embedded Linux. Everything works fine now.
As I mentioned before, I was able to test basic GPIO and SPI.

Hope this helps,
Aurelian

liffiton, 2021-08-04:
Great!  Thanks very much to both of you.

closed 2021-08-04 by liffiton

fanoush, 2021-09-20:
Hi guys, also commented this in pull request #120 that 'fixed' this issue.

> I did not get my project to work when migrated to CM4 from RpiZ. So I examined the CM4 support in WiringPi and found out that CM4 is treated as CM3 and that is not correct in terms of pin numbering afaik. CM4 is more like standard RPI4 etc.

So @toca21 moved from CM4 to Pi Zero and noticed the default pin numbering scheme is **different** on Zero and that was the reason to go back and change CM4 pin numbering scheme so that it is same as Zero? 

and the explanation 
> CM4 is treated as CM3 and that is not correct in terms of pin numbering afaik. CM4 is more like standard RPI4 etc.

feels pretty strange, why then for same reason CM3 would not be 'more like RPI3' or CM1 'more like RPI1'?

The wiringPi pin numbering scheme is extra layer of mapping invented by WiringPi author so that it somehow aligns with 40pin gpio connector present on raspberry boards - see history here http://wiringpi.com/pins/  or here https://projects.drogon.net/wiringpi-pin-numbering/ .  This 40pin connector is not present on any CM module so that is the reason why for all CM modules so far the default is the native Broadcom GPIO pin numbering.

Well, maybe this is all because of CM4 IO board now? Should we take existence of CM4 IO board (or multiple other boards like the waweshare board https://www.waveshare.com/cm4-io-base-b.htm ) as reason for "CM4 is more like standard RPI4 etc." unlike CM3 or CM1?

@toca21 did you actually use CM4 IO board with your project before moving to Zero?

Anyway looks like multiple pin numbering schemes are confusing no matter what default we pick.
Because frankly it is a mess from the start, see description over here http://wiringpi.com/reference/setup/
There are four ways to initialize wiringPi each slightly different.

So I don't know. I'd revert this patch so that CM4 is aligned with all other CM modules however it is true that when the default is the Broadcom numbering the setup methods described at http://wiringpi.com/reference/setup/ are not symmetric and actually don' let you pick up the wiringPi pin numbering scheme at all, as there is no method for it (?). So both calling `wiringPiSetupGpio()` and `wiringPiSetup()` would give you the Broadcom pinout? So this would suggest to actually use the wiringpi numbering as a default **everywhere** because that is the one you can override if you know what you are doing and want to switch to native Broadcom one via `wiringPiSetupGpio()`.