WiringPi/WiringPi #121: Unable of building package with dpkg-buildpackage

DiogoSenico, 2021-07-23:
I was trying to build the package using dpkg-buildpackage and got stuck in an error.
I would like to know if this is this something that is only occurring to me and if this could be fixed.

### How to reproduce:
Run: `dpkg-buildpackage -uc -us -b`

### Logs:
```
make[1]: Leaving directory '/WiringPi-final_official_2.50'
 debian/rules binary
dh binary
dh: warning: Compatibility levels before 10 are deprecated (level 8 in use)
   dh_testroot
   debian/rules override_dh_prep
make[1]: Entering directory '/WiringPi-final_official_2.50'
dh_prep -Xdebian/tmp
make[1]: Leaving directory '/WiringPi-final_official_2.50'
   debian/rules override_dh_auto_install
make[1]: Entering directory '/WiringPi-final_official_2.50'
dh_install
dh_install: warning: Compatibility levels before 10 are deprecated (level 8 in use)
dh_install: warning: Cannot find (any matches for) "debian/tmp/usr/man" (tried in ., debian/tmp)

dh_install: warning: wiringpi missing files: debian/tmp/usr/man
dh_install: error: missing files, aborting
make[1]: *** [debian/rules:40: override_dh_auto_install] Error 255
make[1]: Leaving directory '/WiringPi-final_official_2.50'
make: *** [debian/rules:49: binary] Error 2
dpkg-buildpackage: error: debian/rules binary subprocess returned exit status 2
```

### Possible solution
One solution that I've found was to replace (in file: debian/wiring.install):
this:
```
debian/tmp/usr/bin
debian/tmp/usr/man	usr/share
README.TXT		usr/share/doc/wiringpi
People			usr/share/doc/wiringpi
```
with this:
```
debian/tmp/usr/bin
debian/tmp/usr/share/man	usr/share
README.TXT		        usr/share/doc/wiringpi
People			        usr/share/doc/wiringpi
```

liffiton, 2021-07-26:
Thanks for the detailed report!

I'm not familiar with the Debian packaging side of this, but have you tried `./build debian` using the project's provided build script?  It appears to make a .deb package via a somewhat different route.  It does finish successfully on my test, though I'm not confident in the correctness of the final .deb it creates.

It looks like the debian packaging setup in `debian` hasn't been touched in a very long time, while `debian-template` (which `./build debian` uses) has gotten some more attention.

I could potentially try to fix issues in `debian`, but it might be better to just excise it from the repository and focus on whatever is set up in `debian-template`.

DiogoSenico, 2021-07-27:
Ah yes this works!
I haven't even noticed this possibility, Thank you.

liffiton, 2021-07-27:
Yeah, it's sort of hidden.  Glad it works for you!

I'll leave the issue open, though, since it is still the case that the config under `debian/` just doesn't work.

closed 2023-10-31 by Gadgetoid