WiringPi/WiringPi #159: cannot install wiringpi-2.61-1-armhf.deb on rpi zero w

mpavlikWandera, 2022-03-07:
I am trying to install wiringpi on rpi zero w and getting error about mismatched architecture

```
sudo dpkg -i wiringpi-2.61-1-armhf.deb
dpkg: error processing archive wiringpi-2.61-1-armhf.deb (--install):
 package architecture (arm64) does not match system (armhf)
Errors were encountered while processing:
 wiringpi-2.61-1-armhf.deb

```

Can you please point me in the right direction? I believe I've downloaded the correct package for rpi zero.

Some info about my system
pi@raspberrypi:~/git/WiringPi $ cat /etc/debian_version
11.2
pi@raspberrypi:~/git/WiringPi $ uname -r
5.10.92+
pi@raspberrypi:~/git/WiringPi $ dpkg --print-architecture
armhf

liffiton, 2022-03-07:
Thanks for pointing that out.  Looks like the same file was uploaded for both, and it's only arm64 compatible.

I've removed the "armhf" file from the release page at least, just to keep others from stumbling over the same issue.  I'm not personally set up to build it myself, though, to replace the .deb file with a correct one.

liffiton, 2022-03-08:
I've rebuilt an armhf .deb for 2.61-1 and uploaded it to the release page.  It *should* be all set now, but let me know if it still gives you trouble.

closed 2022-03-08 by liffiton