WiringPi/WiringPi #172: Building version 2.7.0.   The 2.7.0 SO files are not copied by the ./build scrpt to /usr/lib.

mightylastingcode, 2022-09-26:
I wanted to test version 2.70.   I have downloaded the source code.   I ran ./build.   It created the so files (ie libwiringPi.so.2.7.0 in the wiringPi subdirectory under the source code directory).   

The INSTALL document states that this so file is supposed to be copied to the /usr/lib subdirectory and have the soft link (libwiringPi.so) pointed to libwiringPi.so.2.7.0.   

But, it is not so.   I run my example of wiringPiVersion and I still can get 2.70 version.    It does not make sense to me.    How does the compiler find libwiringPi.so.2.70 if it is pointed by the soft link in /usr/lib?

-r--r-- 1 pi pi 72451 Sep 26 14:44 wiringPi.c
-rw-r--r-- 1 pi pi 29540 Sep 26 14:44 wiringPi.o
-rwxr-xr-x 1 pi pi 69936 Sep 26 14:44 libwiringPi.so.2.70
pi@raspberrypi:~/Documents/wiringpi_examples/download/WiringPi/wiringPi $ 

-rwxr-xr-x   1 root root   70236 Mar  8  2022 libwiringPi.so.2.61
-rwxr-xr-x   1 root root   28188 Mar  8  2022 libwiringPiDev.so.2.61
drwxr-xr-x  78 root root   69632 Apr 24 14:59 arm-linux-gnueabihf
lrwxrwxrwx   1 root root      19 Sep 26 14:44 libwiringPi.so -> libwiringPi.so.2.61
lrwxrwxrwx   1 root root      22 Sep 26 14:44 libwiringPiDev.so -> libwiringPiDev.so.2.61
pi@raspberrypi:/usr/lib $

Thank you for any help!
Michael

liffiton, 2022-09-27:
I believe `build` installs the library in `/usr/local/lib` by default, not `/usr/lib`.  Usually, `/usr/local/lib` takes precendence over `/usr/lib`, so that may be why your compiler is finding the new version you built.  The `INSTALL` file doesn't mention `/usr/lib` or a softlink, so I'm not sure what you're referring to there.

mightylastingcode, 2022-09-27:
Thank you for your helpful response.   I found /usr/local/lib.   You are right.  The directory has the new version (2.7.0) of the so files.  This directory is where the gcc linker find the new so files.

You can see that it created two soft links here.  That is what I refer to.

-rwxr-xr-x 1 root root   69852 Sep 26 16:26 libwiringPi.so.2.70
lrwxrwxrwx 1 root root      19 Sep 26 16:26 libwiringPi.so -> libwiringPi.so.2.70  
-rwxr-xr-x 1 root root   27808 Sep 26 16:26 libwiringPiDev.so.2.70
lrwxrwxrwx 1 root root      22 Sep 26 16:26 libwiringPiDev.so -> libwiringPiDev.so.2.70
pi@raspberrypi:/usr/local/lib $ 

-rwxr-xr-x 1 root root 70236 Mar 8 2022 libwiringPi.so.2.61
-rwxr-xr-x 1 root root 28188 Mar 8 2022 libwiringPiDev.so.2.61
lrwxrwxrwx 1 root root 19 Sep 26 14:44 libwiringPi.so -> libwiringPi.so.2.61
lrwxrwxrwx 1 root root 22 Sep 26 14:44 libwiringPiDev.so -> libwiringPiDev.so.2.61
pi@raspberrypi:/usr/lib $

If I unistall the library with ./build uninstall, these links and files will be removed from /usr/local/lib.  Then the gcc linker will look into /usr/lib for the so files.    Because I have installed the previous version (2.6.1) of the library with the dpkg command, the linker will find the previous version (2.6.1) of the so files in /usr/lib.  

Now I understand how it works.   The dpkg install command puts the so files in /usr/lib while the ./build install command puts the so files in /user/local/lib.  When I compile with the gcc toggle_led.c -lwiringPi command, the linker will always check for the so files in /user/local/lib first before /usr/lib.

Is my understanding correct?   Thank you.

Best,
Michael

liffiton, 2022-09-27:
Yup, that sounds right to me!  Generally, directories immediately under `/usr/` like `/usr/lib/` are for things installed by the system package manager (`dpkg` and `apt` in this case), while `/usr/local/` is for things the user installs "manually."

mightylastingcode, 2022-09-27:
Great.  I have a better understanding of this build process.   I would like to contribute to the repos in the future.

I have one more question.  Why all the functions are declared as external in the header file?   

wiringPi.c:   void wiringPiVersion (int *major, int *minor)
wiringPi.h:   extern void wiringPiVersion  (int *major, int *minor) ;

I have tried to add my own function to the library, my linker would complain that MywiringPiVersion cannot be referenced.

wiringPi.c:   void MywiringPiVersion (int *major, int *minor)
wiringPi.h:   extern void MywiringPiVersion  (int *major, int *minor) ;

$gcc toggle_led.c -lwiringPi

Here are the compile errors. 
/usr/bin/ld: /tmp/ccoswOZU.o: in function `main':
toggle_led.c:(.text+0x54): undefined reference to `MywiringPiVersion'
collect2: error: ld returned 1 exit status

Michael

liffiton, 2022-09-27:
That, I'm afraid I don't know.  I did not write the code, and I don't have much experience with developing C/C++ libraries in general.  Sorry.

mightylastingcode, 2022-09-27:
You have been very helpful.   Thank you so much for your replies.  I will open a new post to ask others for their help. 
Michael

closed 2022-09-27 by mightylastingcode