WiringPi/WiringPi #174: Cannot find WiringPi during gcc compile

Aaron3963, 2022-10-11:
I installed WiringPi through ```./build``` and got no problem doing ```gpio -v```. When I try to use it during compilation, I got this:
```
/usr/bin/ld: cannot find -lwiringPi
collect2: error: ld returned 1 exit status
```
I search many methods for fixing this issue, including:

- Uninstall then reinstall
- Add linker to ```/usr/local/lib/```
- Modify ```/etc/ld.so.conf ```

Unfortunately, nothing helped. I am really frustrated and any advice is appreciated.

makbas, 2022-11-07:
A similar problem, old projects using WiringPi not working (compiling) anymore.

sndnvaps, 2022-11-08:
@Aaron3963 @makbas 
i have check the  [wiringPi/Makefile#L96](https://github.com/WiringPi/WiringPi/blob/master/wiringPi/Makefile#L96), when use ./build script to build && insall the wiringPi into system.
you can check 

```makefile
.PHONY:	install
install:	$(DYNAMIC)
	$Q echo "[Install Headers]"
	$Q install -m 0755 -d						$(DESTDIR)$(PREFIX)/include
	$Q install -m 0644 $(HEADERS)					$(DESTDIR)$(PREFIX)/include
	$Q echo "[Install Dynamic Lib]"
	$Q install -m 0755 -d						$(DESTDIR)$(PREFIX)/lib 
	$Q install -m 0755 libwiringPi.so.$(VERSION)			$(DESTDIR)$(PREFIX)/lib/libwiringPi.so.$(VERSION) # it is on the position now ...
	$Q ln -sf $(DESTDIR)$(PREFIX)/lib/libwiringPi.so.$(VERSION)	$(DESTDIR)/lib/libwiringPi.so
	$Q $(LDCONFIG)
```
 if contain  <strong>/usr/local/lib/libwiringPi.so</strong> , <strong>/usr/local/include/wiringPi.h</strong>, you can try add 
this line to makefile 
```makefile
CFLAGS  := -I/usr/local/include -L/usr/local/lib
 gcc -o example $(CFLAGS) example.c -lwiringPi
```

pktmterenceg, 2023-09-11:
Thanks for this response. 

Is this issue still a known/open issue for anyone? 

I'm seeing something _similar_ with `ld`. However, I think it can actually find wiringPi libraries, but for some reason, isn't finding all of the symbols:

`g++ -W -L/usr/local/lib -lwiringPi -lwiringPiDev -li2c -o differential_rpi examples/differential_rpi.c Adafruit_ADS1015.cpp
/usr/bin/ld: /tmp/ccQUM6bi.o: in function `i2cread(unsigned char)':
Adafruit_ADS1015.cpp:(.text+0x1c): undefined reference to `wiringPiI2CRead'
/usr/bin/ld: /tmp/ccQUM6bi.o: in function `i2cwrite(unsigned char, unsigned char)':
Adafruit_ADS1015.cpp:(.text+0x64): undefined reference to `wiringPiI2CWrite'
/usr/bin/ld: /tmp/ccQUM6bi.o: in function `writeRegister(unsigned char, unsigned char, unsigned char, unsigned short)':
Adafruit_ADS1015.cpp:(.text+0xd8): undefined reference to `wiringPiI2CWriteReg16'
/usr/bin/ld: /tmp/ccQUM6bi.o: in function `readRegister(unsigned char, unsigned char, unsigned char)':
Adafruit_ADS1015.cpp:(.text+0x118): undefined reference to `wiringPiI2CWrite'
/usr/bin/ld: Adafruit_ADS1015.cpp:(.text+0x12c): undefined reference to `wiringPiI2CReadReg16'
/usr/bin/ld: /tmp/ccQUM6bi.o: in function `Adafruit_ADS1015::begin()':
Adafruit_ADS1015.cpp:(.text+0x27c): undefined reference to `wiringPiI2CSetup'
/usr/bin/ld: /tmp/ccQUM6bi.o: in function `Adafruit_ADS1015::updateWiringPiSetup()':
Adafruit_ADS1015.cpp:(.text+0x7dc): undefined reference to `wiringPiI2CSetup'
collect2: error: ld returned 1 exit status
pi@raspberrypi:~/Adafruit_ADS1X15_RPi $ `

Thank you in advance.

sndnvaps, 2023-09-12:
@pktmterenceg have you install wiringPi into /usr/lib/

pktmterenceg, 2023-09-12:
Thank you for the response. 

There are symlinks for them in `/usr/lib`:

```
lrwxrwxrwx  1 root root    37 Sep 11 01:16 libwiringPiDev.so -> /usr/local/lib/libwiringPiDev.so.2.70
lrwxrwxrwx  1 root root    34 Sep 11 01:16 libwiringPi.so -> /usr/local/lib/libwiringPi.so.2.70
```

closed 2023-10-31 by Gadgetoid