WiringPi/WiringPi #181: WiringPiI2C.h file no such filr or directory

gauthamikosanam, 2023-06-26:
Hi,
I am working on raspberry pi compute module 4,we are compiling our application in board. In the similar way i want to compile the application in laptop. Cross compiler toolchain installed .
gcc -I/usr/local/include -L/usr/local/lib -s startup/startup.c common/source/*.c -o binaries/tbcuappli.o -lwiringPi------->with this statement it is compiling.
But with below line getting error:
wiringPiI2C.h no such file or directory.
 arm-linux-gnueabihf-gcc -I/usr/local/include -L/usr/local/lib -s startup/startup.c common/source/*.c -o binaries/tbcuappli.o -lwiringPi
I want to compile with arm-linux-gnueabihf-gcc.
Any suggestion?

sndnvaps, 2023-06-29:
you need to cross compile wiringPi first, then use arm-linux-gnueabihf-gcc to link ...

gauthamikosanam, 2023-07-04:
Hi,
I did the wiringPi cross compilation, now .so file was generated for ARM platform.
libwiringPiDev.so.2.50: ELF 32-bit LSB shared object, ARM, EABI5 version 1 (SYSV), dynamically linked, BuildID[sha1]=543074bcfa5e5acb88c8142c5aaa380fc0e9609b, not stripped
libwiringPi.so.2.50:    ELF 32-bit LSB shared object, ARM, EABI5 version 1 (SYSV), dynamically linked, BuildID[sha1]=860dc52199ea7b2781367310a6277477e0b57e99, not stripped
but in my application the wiringPi library I linked with arm-linux-gnueabihf-gcc,getting undefined reference.
/usr/lib/gcc-cross/arm-linux-gnueabihf/9/../../../../arm-linux-gnueabihf/bin/ld: /lib/../lib/libwiringPi.so: undefined reference to `shm_open'
/usr/lib/gcc-cross/arm-linux-gnueabihf/9/../../../../arm-linux-gnueabihf/bin/ld: /lib/../lib/libwiringPi.so: undefined reference to `drcSetupNet'
for this shm_open and drcSetupNet what i need to do?could you please provide any solution for this.

closed 2023-10-31 by Gadgetoid