WiringPi/WiringPi #191: wiringpi-3.0-1.deb

djcleckie, 2024-03-01:
I had V2.70 running OK on a RPi4 on 6.1.216.1.21 
bullseye 64 bit
I have just downloaded wiringpi-3.0-1.deb
Ran GDebi
Seem to work fine
but 
gpio -v is still returning 2.70 

Do I need to completely uninstall V2.70?

Dave

mstroh76, 2024-03-01:
thanks it showed me the problem #192 
you installed the armhf 32-Bit version.
please use arm64 branch for compiling or the prepared arm64 binary from release.
please report if it solved your problem.

djcleckie, 2024-03-02:
Hi

I am quite happy to continue to use V2.70 on my RPi 4 until such time as a 64 bit version is avaiable.
The reason I tried V3.0-1 was that I quite fancy a RPi 5.

The only reason I am running the 64 bit version is that software is starting to appear that is 64 bit only.

Thanks for the update
Dave

mwallner, 2024-03-02:
Hey @djcleckie ,- thanks for bringing this up!
@mstroh76 added the 64bit binaries to the [release page](https://github.com/WiringPi/WiringPi/releases/tag/3.0)

closed 2024-03-02 by mwallner