WiringPi/WiringPi #194: "Unable to determine board revision from /proc/cpuinfo" on none Raspberry Pi OS

mstroh76, 2024-03-08:


pasquale90, 2024-03-08:
Hello there. 

I can;t be sure if this is adequate, but I managed to resolve this using https://github.com/WiringPi/WiringPi/issues/108#issuecomment-817384407. I also replaced the line

*model = PI_MODEL_4B ; *rev = PI_VERSION_1_2 ; *mem = 2 ; *maker = PI_MAKER_EGOMAN;

with

*model = PI_MODEL_3B ; *rev = PI_VERSION_1_2 ; *mem = 2 ; *maker = PI_MAKER_EGOMAN;

...., as I use a rpi 3B. After changing the file, I recompiled and the runtime error disappeared.

Informatively, I am using debian server on a raspberry pi 3 . 

```
$ uname -a
Linux rpi3-20231109 6.1.0-13-arm64 #1 SMP Debian 6.1.55-1 (2023-09-29) aarch64 GNU/Linux

$ cat /proc/cpuinfo

processor       : 0
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

```

mstroh76, 2024-03-08:
please execute https://github.com/GrazerComputerClub/Benchmark/blob/master/src/pidetect.py and 'lsb_release -a' and post the output

pasquale90, 2024-03-08:
```
$ python3 pidetect.py
model: Raspberry Pi 3 Model B Rev 1.2, modelno: 8, RAM: 1024 MB(2), revision: 0xa02082, serial: 0x9eebd2e9

$ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
```

mstroh76, 2024-03-08:
I will fix it this weekend. Be prepared for testing...

closed 2024-03-10 by mwallner

MuzammilPatel, 2025-08-26:
Hi @mstroh76 , I am facing the same issue on a Pi5. 

` $ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
`

`$ python pidetect.py
model: Raspberry Pi 5 Model B Rev 1.0, modelno: 23, RAM: 8192 MB(5), revision: 0xd04170, serial: 0xf89b5888f4ed27c6
`

`$ cat /proc/cpuinfo
processor       : 0
BogoMIPS        : 108.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x4
CPU part        : 0xd0b
CPU revision    : 1

processor       : 1
BogoMIPS        : 108.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x4
CPU part        : 0xd0b
CPU revision    : 1

processor       : 2
BogoMIPS        : 108.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x4
CPU part        : 0xd0b
CPU revision    : 1

processor       : 3
BogoMIPS        : 108.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x4
CPU part        : 0xd0b
CPU revision    : 1

Revision        : d04170
Serial          : f89b5888f4ed27c6
Model           : Raspberry Pi 5 Model B Rev 1.0
`
How do you resolve

PR #196 (2024-03-10): Revision
closes #194 
closes #195 
bumps version to 3.1
cleanup bash scripts