WiringPi/WiringPi #211: pinMode PWM: Unable to do this when using /dev/gpiomem. Try sudo?

Owhenthesaints, 2024-03-29:
I have been scouring the web for a few hours now and I cannot seem to fix the problem. I am using rpi with ubuntu 22.04 server
`pinMode PWM: Unable to do this when using /dev/gpiomem. Try sudo?`
I am running my executable through ROS2 and hence cannot give it root privileges. I have already tried `sudo adduser pi gpio`, `sudo usermod -a -G gpio $USER`, `sudo chown root.gpio /dev/gpiomem` and`sudo chmod g+rw /dev/gpiomem` and none will do. Thanking you for your answer if there is one.

Owhenthesaints, 2024-03-29:
sorry just reread your description

mstroh76, 2024-03-29:
What Wiring Pi Version do you use?
It's sounds like you need /dev/mem to use the feature, so changing rights for /dev/gpiomem will not help.

Commend in Source: 
_See if we're using the /dev/gpiomem interface, if-so then some operations can't be done and will crash the Pi._

Looks like it's not possible without root access. If I find the time I can test if it's really crashing (need to bypass the check). 
Maybe PWM its outside of gpiomem memory, like at Pi5, then there is no solution to this problem.

Owhenthesaints, 2024-03-29:
I am using version 3.0

Owhenthesaints, 2024-03-29:
i have also changed the rights of /dev/mem

Owhenthesaints, 2024-03-31:
If it is not doable it is fine I have found a solution to go sudo with ros2. I saw this question pop up on other forums so I think it would be a good idea if it is not doable to just put it down. I would investigate myself but I don't really have the know how yet.

mstroh76, 2024-04-02:
Thanks for the issue.
I didn't even know how it actually worked.

**Analysis so far:**
Usually /dev/mem is accessed directly, even from a non-root account.
This works for memory areas intended for mapped device address areas. However, only if this was activated in the kernel (compiler option).

If this is not possible WiringPi use /dev/gpiomem (kernel module). However, the addresses for PWM are not available on that memory (only GPIO)! That's why PWM doesn't work. `pinMode PWM: Unable to do this when using /dev/gpiomem`

**Actions:**
I will add an additional detection in the next version so that it is clear whether /dev/mem or /devgpiomem is used.
I will retext the error message so that it is clear why the error occurs.
We will do further tests with Ubuntu (I suspect this is where the problem occurs, because of Kernel settings)

*Solution:*
So the solution for you is to ensure that a normal user can access /dev/mem (device addresses).
You may need a new kernel for this.

Please keep me updated if you were able to solve the problem.

mstroh76, 2024-04-12:
#219 a solution too

mstroh76, 2024-04-14:
report for analysis finished #221