WiringPi/WiringPi #212: i2c write sizes other than 8/16

konsumer, 2024-04-01:
I may just not understand how to do it correctly, but I can't figure out how to write 4 bytes on i2c. I am trying to port [this](https://github.com/konsumer/8encoder/blob/main/M58Encoder.py) to WiringPi. I got most of them working, but color for RGB LEDs is `[register + (index*3), r, g, b]`. I think this would work, if I had access to `i2c_smbus_access` (although I am not sure what `command` is.)

```c
void wiringpi_8encoder_set_led_color_rgb(uint8_t fd, uint8_t index, uint8_t r, uint8_t g, uint8_t b) {
  uint8_t v4[4];
  v4[0] = (index * 3) + RGB_LED_REG;
  v4[1] = r;
  v4[2] = g;
  v4[3] = b;

  // what is this?
  uint8_t command = 0;

  i2c_smbus_access(fd, I2C_SMBUS_WRITE, command, 4, &v4);
}
```

Is this related to #130 ?
Would it make sense to expose a "more raw" set of wrapper functions, or just directly expose `i2c_smbus_access`?
What is `command`?

mstroh76, 2024-04-02:
All clear, i know what you need/want. Can be included soon.
No need for PR,  unless you want to do it.

konsumer, 2024-04-02:
I tried using the code in #130 PR, and it did not seem to work, at least for my case.

I ended up going with pigpio, and it's implemented [here](https://github.com/konsumer/8encoder/blob/main/pigpio-8encoder/pigpio-8encoder.h). They have some helpers, but it seemed to work better (and match my python) to just use the raw read/write byte functions.

mstroh76, 2024-04-04:
we lost you to pigpio 😱 , the second best library.

I ended up implementing directly a long time ago -> https://github.com/mstroh76/RaspberryPiProjekte/blob/master/Elektronik/HP03S.c

raw read/write byte functions are very likely to be included in future releases of WiringPi

konsumer, 2024-04-05:
> we lost you to pigpio 😱 , the second best library.

I am not totally sold on pigpio, and I still like a lot of things about wiringpi. Pigpio seems slow, and I don't like the mix of system-service and stuff, for gpio. I want it to just be fast and work reliably, without loading a lot of other layers.

The fastest & simplest code for working with these rotaries I have tried far has been python, which really surprises me, but is not really an option for a native puredata extension.

I started working on [this](https://github.com/konsumer/RasPiArduino). I really like the idea of the same (or very similar) code working in multiple places (like arduino.) It's something blinka/circuitpython does really well. It's awesome to find a driver lib for circuitpython (maybe originally written for ESP32 or whatever) and just drop it into a pi project and have it work. I'd like something similar for C/C++. For this particular project, I have 3 i2c devices that all have arduino libs for them, and they can essentially be used as-is, once the infrastructure code is in place.

> I ended up implementing directly a long time ago

ooh, this looks good, actually. I might end up using this in my immediate project.

konsumer, 2024-04-06:
For a follow-up: I made [this](https://github.com/konsumer/8encoder/tree/main/linux-8encoder) using regular  smbus stuff, and it works great! I still need to work out the puredata part for it, and then make similar for other 2 devices, but this actually works perfect for my needs, and ended up really simple.

This is what I went with:

```c
// Get a value, by register
bool i2c_get_register_val(int i2c, uint8_t reg, void* out_pntr, uint8_t len) {
  if (write(i2c, &reg, 1) != 1) {
    return false;
  }
  // 1us pause helps with missing messages
  usleep(1);
  if (read(i2c, out_pntr, len) != len) {
    return false;
  }
  // 1us pause helps with missing messages
  usleep(1);
  return true;
}

// Set a value, by register (max length 32 bytes)
bool i2c_set_register_val(int i2c, uint8_t reg, void* in_ptr, uint8_t len) {
  uint8_t msg[33] = {0};
  msg[0] = reg;
  memcpy(msg+1, in_ptr, len);
  
  if (write(i2c, msg, len+1) != (len+1)) {
    return false;
  }

  // 1us pause helps with missing messages
  usleep(1);
  return true;
}

// Example Usage

typedef struct {
  uint8_t r;
  uint8_t g;
  uint8_t b;
} ColorRGB;

void linux_8encoder_set_led_rgb(int8_t i2c, uint8_t index, ColorRGB value) {
  i2c_set_register_val(i2c, RGB_LED_REG + (index * 3), &value, 3);
}

ColorRGB linux_8encoder_get_led_rgb(int8_t i2c, uint8_t index) {
  ColorRGB out = {0};
  i2c_get_register_val(i2c, RGB_LED_REG + (index * 3), &out, 3);
  return out;
}
```

mstroh76, 2024-04-28:
05248ae7e61a546b99269fc3039e9ef90ac94e20 now implemented

closed 2024-04-28 by mstroh76