WiringPi/WiringPi #216: shared object versioning

rafael2k, 2024-04-05:
Right now in "deb-packaging" branch repo I'm setting so name to 3.2, so we get: 

```
/usr/lib/aarch64-linux-gnu/libwiringPiDev.so -> libwiringPiDev.so.3.2
/usr/lib/aarch64-linux-gnu/libwiringPiDev.so.3 -> libwiringPiDev.so.3.2
/usr/lib/aarch64-linux-gnu/libwiringPiDev.so.3.2
/usr/lib/aarch64-linux-gnu/libwiringPi.so -> libwiringPi.so.3.2
/usr/lib/aarch64-linux-gnu/libwiringPi.so.3 -> libwiringPi.so.3.2
/usr/lib/aarch64-linux-gnu/libwiringPi.so.3.2
```
```
ldd /usr/bin/gpio
        linux-vdso.so.1 (0x0000007f979ee000)
        libwiringPi.so.3 => /lib/aarch64-linux-gnu/libwiringPi.so.3 (0x0000007f97940000)
        libwiringPiDev.so.3 => /lib/aarch64-linux-gnu/libwiringPiDev.so.3 (0x0000007f97910000)
```

The idea is to keep 3 as major version until ABI breaks, right?

mwallner, 2024-04-12:
in short: yes 
We've moved from v2 to v3 more to emphasize picking up the ball rather than doing breaking changes - in theory v4 would have breaking changes - which is not what we're looking for :wink: 
So unless something unforeseen happens, v3 is here to stay :tada:

rafael2k, 2024-04-12:
Thanks @mwallner !
So we keep like this.
: )

closed 2024-04-19 by mwallner