WiringPi/WiringPi #218: Error in openSUSE tumbleweed，4B

Dregen-Yor, 2024-04-11:
When I try to use gpio readall, it return wiringPiSetup: mmap (GPIO) failed: Operation not permitted.

Dregen-Yor, 2024-04-12:
In addition, I do it as the root.

mwallner, 2024-04-12:
that's interesting, could you please provide the output when you do the following sequence of commands?

```
su
who
WIRINGPI_DEBUG=1 gpio readall
```

mwallner, 2024-04-12:
actually might be related to #211
your kernel configuration might not support DMA for gpio access, so you might have to fall back to libgpiod until we'll have support for the userspace mode access types in WiringPi (on our list, but cannot give you any due dates)

Dregen-Yor, 2024-04-13:
> that's interesting, could you please provide the output when you do the following sequence of commands?
> 
> ```
> uname -a
> su
> who
> WIRINGPI_DEBUG=1 gpio readall
> ```

The output is 
[image]

Dregen-Yor, 2024-04-13:
what can I do to solve this problem, please help.  qwq

mstroh76, 2024-04-13:
Do you have device 
/dev/gpiomem ?

Dregen-Yor, 2024-04-14:
> Do you have device 
> /dev/gpiomem ?

no, but I have /dev/gpiochip0 and /gpiochip1

mstroh76, 2024-04-15:
Wait for #219 or change kernel with DMA support.

mstroh76, 2024-04-16:
If you're brave, you can test branch GPIOAPI with the new kernel GPIO device implementation. its Slow and only basic testing is done so far.

Need to call ``wiringPiSetupGpioDevice()`` to activate

mwallner, 2024-04-19:
@mstroh76 - would the user always have to call `wiringPiSetupGpioDevice`, or will this be done automatically as a fallback (when no DMA support is detected)?

mstroh76, 2024-05-08:
'Support Kernel Character Device interface' is now included since version 3.4 .
It's only used if wiringPiSetupGpioDevice is called from your program. There is no fallback.
'gpio readall' is not working with new Kernel interface, only if DMA is possible.