WiringPi/WiringPi #220: Sensor READ fails on even number of reads

sgreensides, 2024-04-12:
CM4 with latest OS and WiringPi - clean install of both
Same circuit with python works every-time = so assumed to be correct
Current work-around is to check the size of return and call again if too low

C++ Code
=======
  class HCSR04 {
  public:
    HCSR04 (const uint8_t pEchoPin, const uint8_t pTriggerPin) {
      this->_EchoPin = pEchoPin;
      this->_TriggerPin = pTriggerPin;
      pinMode(this->_TriggerPin, OUTPUT);
      pinMode(this->_EchoPin, INPUT);
      digitalWrite(this->_TriggerPin, LOW);
    }                                                           // end constructor

   /*
    * get the distance reading - RAW
    * and return it
    */
   uint64_t getDistance (uint64_t pTimeOut = 50000) {
     //
     // blip the trigger pin
     // first we set it LOW - so that we are starting from a known state
     // blip = set HIGH
     // set LOW to end
     //
     digitalWrite(this->_TriggerPin, LOW);
     delayMicroseconds(100);                                        // time for GPIO pin to settle ? needed ?
     // known state now

     // so blip
     digitalWrite(this->_TriggerPin, HIGH);
     delayMicroseconds(10);                                        // check whether we can reduce this!!
     digitalWrite(this->_TriggerPin, LOW);

     // wait for echo to return
     // pTimeOut is unsigned so should never be allowed to approach zero
     while ((digitalRead(this->_EchoPin) == LOW) && (pTimeOut-- > 2)) {
       // or fail on timeout
       ;
     };                                                             // end waiting for echo;

     // record this time
     this->_startTimeUsec = micros();

     // whilst we have an echo
     while ( digitalRead(this->_EchoPin) == HIGH ) {
       // wait for the ECHO to finish
       ;
     };                                                         // end of whilst echo echo exists

     // difference between now and when we first started
     return (micros() - this->_startTimeUsec);
   }

  /*****/
  private:
    uint8_t _EchoPin;
    uint8_t _TriggerPin;
    uint64_t _startTimeUsec;
};                                                              // end of class HCSR04

sgreensides, 2024-04-12:
Apologies if not pasted code correctly = not done this before :)

mstroh76, 2024-05-08:
This is your reference implementation (and WiringPi test case) :
https://github.com/GrazerComputerClub/Sensors-WiringPi/tree/master/HC-SR04
please check if its working and compare to your implementation