WiringPi/WiringPi #223: Failed to compile wiringpi

pedross3, 2024-04-15:
Hi everyone!

When installing wiringPi, I faced the issue that my raspberry wasn't being able to install a.deb package via command line. Then I just double clicked it, and successfully installed.
But when I try to compile it, it doesn't fiind it's directory.

`/usr/bin/ld: cannot find -lwiringpi: No such file or directory
collect2: error: ld returned 1 exit status`

So I want to ask for your help to fix this issue.
Thank you very much!

pedross3, 2024-04-15:
> Need Output:
> 
> ```
> apt show wiringpi
> lsb_release -a
> gpio -v
> ls -l /usr/lib/libw*
> ```

apt show wiringpi

Package: wiringpi
Version: 3.2
Status: install ok installed
Priority: optional
Section: libraries
Maintainer: Grazer Computer Club - GC2 <wiringpi@gc2.at>
Bugs: https://github.com/WiringPi/WiringPi/issues
Installed-Size: unknown
Depends: libc6
Homepage: https://github.com/WiringPi/WiringPi
Uploaders: Grazer Computer Club - GC2 <wiringpi@gc2.at>
Download-Size: unknown
APT-Manual-Installed: yes
APT-Sources: /var/lib/dpkg/status
Description: The wiringPi libraries, headers and gpio command
  Libraries to allow GPIO access on a Raspberry Pi from C and C++
  programs as well as from the command-line

lsb_release -a

No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm

gpio -v

gpio version: 3.2
Copyright (c) 2012-2024 Gordon Henderson and contributors
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Pi 4B, Revision: 01, Memory: 1024MB, Maker: Sony 
  * Device tree is enabled.
  *--> Raspberry Pi 4 Model B Rev 1.1
  * This Raspberry Pi supports user-level GPIO access.

ls -l /usr/lib/libw

ls: cannot access '/usr/lib/libw': No such file or directory

mstroh76, 2024-04-15:
did you forget * ?

pedross3, 2024-04-15:
> did you forget * ?
> 
> `ls -l /usr/lib/libw*`

ls -l /usr/lib/libw*
lrwxrwxrwx 1 root root     21 Apr 15 09:16 /usr/lib/libwiringPiDev.so -> libwiringPiDev.so.3.2
-rwxr-xr-x 1 root root  77864 Apr 12 13:45 /usr/lib/libwiringPiDev.so.3.2
lrwxrwxrwx 1 root root     18 Apr 15 09:16 /usr/lib/libwiringPi.so -> libwiringPi.so.3.2
-rwxr-xr-x 1 root root 157432 Apr 12 13:45 /usr/lib/libwiringPi.so.3.2

mstroh76, 2024-04-15:
``
ldd /usr/bin/gpio
ls -l /lib/libw*
file /lib/libwiringPi.so.3.2
``

pedross3, 2024-04-15:
> ```
> ldd /usr/bin/gpio
> ls -l /lib/libw*
> file /lib/libwiringPi.so.3.2
> ```
> 
> 32-Bit or 64-Bit system?

ldd /usr/bin/gpio
	linux-vdso.so.1 (0x0000007fa0bf2000)
	libwiringPi.so => /lib/libwiringPi.so (0x0000007fa0b50000)
	libwiringPiDev.so => /lib/libwiringPiDev.so (0x0000007fa0b20000)
	libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000007fa0970000)
	libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000007fa08d0000)
	libcrypt.so.1 => /lib/aarch64-linux-gnu/libcrypt.so.1 (0x0000007fa0880000)
	/lib/ld-linux-aarch64.so.1 (0x0000007fa0bb5000)

ls -l /lib/libw*
lrwxrwxrwx 1 root root     21 Apr 15 09:16 /lib/libwiringPiDev.so -> libwiringPiDev.so.3.2
-rwxr-xr-x 1 root root  77864 Apr 12 13:45 /lib/libwiringPiDev.so.3.2
lrwxrwxrwx 1 root root     18 Apr 15 09:16 /lib/libwiringPi.so -> libwiringPi.so.3.2
-rwxr-xr-x 1 root root 157432 Apr 12 13:45 /lib/libwiringPi.so.3.2

file /lib/libwiringPi.so.3.2
/lib/libwiringPi.so.3.2: ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, BuildID[sha1]=ef08c2bbc99eb38d144153ef6ef40fbad742bc4f, not stripped

64-bit

mstroh76, 2024-04-15:
```
sudo ldconfig  
ls -l /usr/include/wiring*
```
retry 

**I have no idea**

aurelian17, 2024-04-15:
Can you please share how your project links to wiringpi?
Can you please check the LD_LIBRARY_PATH? I'm saying this because from the error reported, looks like the linker doesn't "see" the wiringpi library.

pedross3, 2024-04-15:
> Can you please share how your project links to wiringpi? Can you please check the LD_LIBRARY_PATH? I'm saying this because from the error reported, looks like the linker doesn't "see" the wiringpi library.

g++ -Wall -lwiringpi
how do I check LD_LIBRARY_PATh?

> 
> And please try exporting the path to wiringpi library and then build again with the following command: export LD_LIBRARY_PATH="/lib:/usr/lib:"

Please clarify this step.

aurelian17, 2024-04-15:
> > Can you please share how your project links to wiringpi? Can you please check the LD_LIBRARY_PATH? I'm saying this because from the error reported, looks like the linker doesn't "see" the wiringpi library.
> 
> g++ -Wall -lwiringpi how do I check LD_LIBRARY_PATh?
> 
Ok, this is the trick, you have no environment setup, so this is why the linker doesn't detect the wiringpi library.

You can check the LD_LIBRARY_PATH with echo ${LD_LIBRARY_PATH}, but I assume is empty...

> > And please try exporting the path to wiringpi library and then build again with the following command: export LD_LIBRARY_PATH="/lib:/usr/lib:"
> 
> Please clarify this step.
Do the following:
export LD_LIBRARY_PATH="/lib:/usr/lib:"
g++ -Wall -lwiringpi ...

pedross3, 2024-04-15:
Correct, it is empty.

aurelian17, 2024-04-15:
> Correct, it is empty.

Have you tried compiling with the LD_LIBRARY_PATH exported like in my previous comment?

pedross3, 2024-04-16:
> > Correct, it is empty.
> 
> Have you tried compiling with the LD_LIBRARY_PATH exported like in my previous comment?

g++ -Wall -o \ -L /home/pesan/WiringPi -lwiringpi
did this but I'm not sure if the syntax is working. I've looked up on StackOverflow (https://stackoverflow.com/questions/1904990/what-is-the-difference-between-ld-library-path-and-l-at-link-time) and ran ldconfig -v on the terminal to make sure the linker recognizes wiringpi. 
It is there indeed.

aurelian17, 2024-04-16:
I don't understand what are you trying to do. Are you trying to build a file .c or .cpp that uses the wiringpi? If yes, please share that file, or at least a simple version of it and the full command you are trying to compile it with. Why do you pass -L /home/pesan/WiringPi ? Why not use the default path where the wiringpi so is located? What does the /home/pesan/WiringPi contain?

pedross3, 2024-04-16:
You won't believe what was the problem.

I missed the capital P when `-lwiringPi`

closed 2024-04-16 by pedross3