WiringPi/WiringPi #226: wiringPiSPISetup() missing close(fd) function

jpitz31, 2024-04-30:
Normally I only open an SPI channel when I need it,  after performing a series of digitalRead() or digitalWrite()'s I then close the connection.  

Not sure, but did there use to be an SPI close(fd) function.  I did some searching in the source and found other interfaces do use the
close(fd) function.

Not sure if it was deprecated or not, for SPI.

 // enable SPI
 int channel = 0; // Use SPI channel 0
 int speed = 1000000; // Set SPI clock speed to 1 MHz
 int File_Desc = wiringPiSPISetup(channel, speed);

In this case, if there is no close(fd)  then I can only call the wiringPiSPISetup()  one time, otherwise I run out of File Descriptors.

Thanks for taking over the project.

Cheers

Joe

mwallner, 2024-04-30:
Unfortunately cleanup routines are not implemented as of now, this should not be a problem unless you're dealing with an application that's called millions of times - if you're only running a program once in a while that itself uses WiringPi you'll be fine.

Thanks for bringing it to our attention - we'll do something about it in future releases.

For now, we'll be like pirates:
> Take what you can, give nothing back

mstroh76, 2024-05-13:
resolved with [5546986](https://github.com/WiringPi/WiringPi/commit/55469861299ae48e19e1b1d8d25f5b3b7d8ebd9a)

closed 2024-05-13 by mstroh76