WiringPi/WiringPi #243: Add optional parameter to spi to select spidev

LaneaLucy, 2024-05-10:
https://github.com/WiringPi/WiringPi/blob/8960cc91b911db8ec0c272781edf34b8aedb60d9/wiringPi/wiringPiSPI.c#L112

Spidev0 is hard coded, but for example the devterm with cm4 uses spidev4 for its printer. In the past, the creators of devterm offered a custom version of wiringpi because of this, but if there would be a optional parameter to the spi functions, to select different spidev's, then you could use a unmodified version of wiringpi.
The modification needed for this shouldn't be that much of work, first remove the hardcoded spidev and make it also a variable, like the channel, get the var from optional function parameter, and last, make the spiFds array bigger or better, make it a two dimensional array.

This would really help in porting a new os to the Devterm without making custom wiringpi versions for everything.

Thx in advance

mstroh76, 2024-05-11:
I will try to resolve this weekend.

LaneaLucy, 2024-05-11:
Thank you very much and have a nice weekend

LaneaLucy, 2024-05-12:
Could you please make a deb out of it so it's easier to test?

Thx in advance

LaneaLucy, 2024-05-14:
Ignore my last message.

Build it today and it works great.
Thank you very much

closed 2024-05-14 by LaneaLucy