WiringPi/WiringPi #246: Installation procedure (e.g. for .sh files)

dsyleixa, 2024-05-16:
how is the complete installation procedure for the new 3.4 version?
e.g. for the old version it once was
[quote="old one"]
```
[code]
git clone https://github.com/WiringPi/WiringPi
cd WiringPi
./build debian
ls debian-template/*deb
[/code]
```
[/quote]?
which is the full deletion code to remove all and everything from an old version before, not letting behind any orphans or File corpses?

mstroh76, 2024-05-18:
To reinstall the WiringPi library, actually there is no removing the old version needed, because nothing has changes so far. 
But do be save, you can first remove the old version files using `sudo apt purge -y wiringpi`. Then download the latest version from the official GitHub repository. Afterwards install it with `sudo apt install ./wiringpi_3.4_armhf.deb`. Finally, verify the installation by running `gpio -v` and `gpio readall`.

By the way I'm using the script [reinstall](https://github.com/WiringPi/WiringPi/blob/master/reinstall)

closed 2024-06-02 by mstroh76

dsyleixa, 2024-09-17:
how to remove an old version which is not from a apt package but from former gitclone versions?