WiringPi/WiringPi #251: Wiring PI on Void Linux ...

mallory98e, 2024-05-31:
```
-bash-5.2$ ./xbps-src pkg WiringPi
=> xbps-src: updating repositories for host (aarch64)...
[*] Updating repository `https://repo-default.voidlinux.org/current/aarch64/bootstrap/aarch64-repodata' ...
[*] Updating repository `https://repo-default.voidlinux.org/current/aarch64/aarch64-repodata' ...
[*] Updating repository `https://repo-default.voidlinux.org/current/aarch64/nonfree/aarch64-repodata' ...
[*] Updating repository `https://repo-default.voidlinux.org/current/aarch64/debug/aarch64-repodata' ...
=> xbps-src: updating software in / masterdir...
=> xbps-src: cleaning up / masterdir...
=> WiringPi-0.0.20200408_3: removing autodeps, please wait...
=> ERROR: WiringPi-0.0.20200408_3: this package cannot be built for aarch64.
```

I looked into that build script coming with this, it would never work as it's using Debian based scripts inside.

Only thing I can find is some python garbage:

[-] python3-RPi.GPIO-0.7.1_3 Module to control Raspberry Pi GPIO channels

I really don't want to switch to other utility than WiringPI so is there a way to compile it on Void Linux?!

mallory98e, 2024-06-04:
jesus i wonder why did they needed to scrap a completely good project, wiring pi worth more than 100 systemDs alltogether :+1:

mstroh76, 2024-06-05:
"We do not officially support Void Linux and the XBPS Package Manager. However, you can try installing it using the following steps:
  ```
  git clone https://github.com/wiringpi/wiringpi
  ./build
  ```
Please note that we do not recommend this method, as we prefer using Debian packages.

mallory98e, 2024-06-06:
Wow this worked!

But is it really working for Raspberry PI 5? ;)

```
 +-----+-----+---------+------+---+---Pi 5---+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 |   -  | 0 |  3 || 4  |   |      | 5v      |     |     |
 |   3 |   9 |   SCL.1 |   -  | 0 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   -  | 0 |  7 || 8  | 0 |  -   | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 0 |  -   | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 |   -  | 0 | 11 || 12 | 0 |  -   | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 |   -  | 0 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 |   -  | 0 | 15 || 16 | 0 |  -   | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 0 |  -   | GPIO. 5 | 5   | 24  |
 |  10 |  12 |    MOSI |   -  | 0 | 19 || 20 |   |      | 0v      |     |     |
 |   9 |  13 |    MISO |   -  | 0 | 21 || 22 | 0 |  -   | GPIO. 6 | 6   | 25  |
 |  11 |  14 |    SCLK |   -  | 0 | 23 || 24 | 0 |  -   | CE0     | 10  | 8   |
 |     |     |      0v |      |   | 25 || 26 | 0 |  -   | CE1     | 11  | 7   |
 |   0 |  30 |   SDA.0 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.0   | 31  | 1   |
 |   5 |  21 | GPIO.21 |   -  | 0 | 29 || 30 |   |      | 0v      |     |     |
 |   6 |  22 | GPIO.22 |   -  | 0 | 31 || 32 | 0 |  -   | GPIO.26 | 26  | 12  |
 |  13 |  23 | GPIO.23 |   -  | 0 | 33 || 34 |   |      | 0v      |     |     |
 |  19 |  24 | GPIO.24 |   -  | 0 | 35 || 36 | 0 |  -   | GPIO.27 | 27  | 16  |
 |  26 |  25 | GPIO.25 |   -  | 0 | 37 || 38 | 0 |  -   | GPIO.28 | 28  | 20  |
 |     |     |      0v |      |   | 39 || 40 | 0 |  -   | GPIO.29 | 29  | 21  |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+---Pi 5---+---+------+---------+-----+-----+

```

It seems that next to IN and OUT no ALT modes can be set on the pins :(

  ./gpio/readall.c:  "IN", "OUT", "ALT5", "ALT4", "ALT0", "ALT1", "ALT2", "ALT3", "ALT6", "ALT7", "ALT8", "ALT9"

Wonder why that is.

For example I liked to use this to reclaim the hardware UART:

  gpio mode 15 ALT0; gpio mode 16 ALT0

mstroh76, 2024-06-07:
_pinModeAlt:  *	This is an un-documented special to let you set any pin to any mode_
RETURN_ON_MODEL5  -> *we are not supporting by Pi 5 because of unknown results/problems/consequences** 

simply remove the line if you like
https://github.com/WiringPi/WiringPi/blob/813c3130be372841d84653dc8a1c2bf6e500d0e2/wiringPi/wiringPi.c#L1618

Normaly devicetree and rasp-config are used for activating interfaces like i2c, spi and uart.

mstroh76, 2024-06-07:
You should refer to ‘Table 4: GPIO Function Selection’ in the RP1 peripherals document at [this link](https://datasheets.raspberrypi.com/rp1/rp1-peripherals.pdf). However, keep in mind that this table only covers GPIO functions (A4 for UART on pin14&15) . For baud rate settings and other configurations, as well as information about the Linux serial device, you’ll need additional resources. **I recommend using an operating system with Raspberry Pi support.**