WiringPi/WiringPi #264: improvement for gpio wfi possible?

Joerg12345, 2024-07-17:
Thanks for the work translating WiringPi for the Pi5 !

Is it possible that the command gpio ... wfi ... returns the status of the GPIO (0 or 1) when the interrupt occurs ?

For example, I use gpio -g wfi <gpioNumber> both.
After the interrupt occurs, I then have to re-read the GPIO with the command gpio -g read <gpioNumber> to find out whether the interrupt occurred at falling (0) or rising (1).
If gpio wfi returned the status at which the interrupt occurred, I would no longer have to re-read the GPIO.

Thank you!
Jörg

mstroh76, 2024-07-17:
Yes its possible and useful
I had the same problem, but with the Library ISR Function
The new ISR reports the level after IRQ, but it is not implemented so far.
So we need new/advanced ISR implementation, new API call, new parameter (wifs) for gpio tool.

phylax2020, 2025-02-18:
I'll  try to enhance the function waitForInterrupt in the branch GPIO_V2_IOCTL to return a struct containing data derived from
 struct gpio_v2_line_event included in <linux/gpio.h> \:

The struct could look something like this:

status \: -1 error; 0 timeout; >0 success
edge  \: INT_EDGE_FALLING or INT_EDGE_RISING
timestamp \:  in microseconds, when IRQ happened

timestamp is usefull to measure times between events.

mstroh76, 2025-05-16:
gpio wfis wfis <pin> <edge> <debounce period> <iterations> <timeout>

Exit code:
1 ... Falling/LOW
2 ... Rising/High
-1/255 ... Timeout
-2/254 ... Wrong parameter

closed 2025-05-16 by mstroh76