WiringPi/WiringPi #292: Does this library support cross-build?

duanju, 2024-10-21:
I am developing an application in my ubuntu pc with x64 architecture. I tried to build my application with command: _aarch64-linux-gnu-gcc -o driver main.c motor.c -lwiringPi_. Several errors ocurred when i compile my application.

error 1:
motor.h:5:10: fatal error: wiringPi.h: No such file or directory
error 2:
/usr/lib/gcc-cross/aarch64-linux-gnu/12/../../../../aarch64-linux-gnu/bin/ld: warning: libcrypt.so.1, needed by libs/libwiringPi.so, not found (try using -rpath or -rpath-link)

Is it possible to develop app in x64 machine? Thank you.

sam-vdp, 2025-02-12:
For me, the linker error was caused by an outdated glibc in the build vm.
Updating the build environment to Debian bookworm fixed the issue for me.

Stetsed, 2025-04-06:
I can confirm I am currently running into the same issue, I was able to get the wiringPi to be added to my project by instead of having it compile itself, just copying over the WiringPi and WiringPiD directories to avoid any issues there. However it also requires crypt to be linked which seems like an intenrly diffrent headache.

If anybody has found a way to do this without too much issue I would also love to hear it, and if I end up figuring out an way that's simple I will publish here :D