WiringPi/WiringPi #293: Possibility to adjust SPI read-write byte count dynamically

linuxchris12, 2024-10-22:
There are some SPI devices which require dynamic rw calls, meaning that: before the rw transaction is initiated, it is not clear how many bytes will be returned. The length is determined within the first couple of bytes only.

Example:

RPi sends 2 bytes, a read all data command
Device replies 83 bytes, first two bytes indicating the lenght of the payload, here 81 bytes.

That is all within a single rw command. We don't know the amount of bytes which is returned beforehand.
This kind of command is usually supported by the hardware, but it seems like it is not possible using WiringPi.

I had to switch to a differenent library in the past to support this. But since the release of RPi 5, I am considering using WiringPi again. 

Is there any chance you can extend the library to contain such a feature?

mstroh76, 2024-10-27:
Hi, if you look at the function 'wiringPiSPIxDataRW,' you'll see how easy it is to implement SPI transfer. I suggest that you implement it in your program. Why do you think it needs to be inside the library?

linuxchris12, 2024-11-08:
Hi,

thank you for sharing this. I can see that the wiringPiSPIxDataRW function does not allow to control the chip select pin directly, which would be required. The problem is that the chip select pin will be cleared automatically after each usage of wiringPiSPIxDataRW, which will terminate the SPI transfer. If that can be prevented, you could use that function indeed as you suggest. But like it is right now, that is not possible. It could be a limitation from the OS or even the new RP1 chip though...