WiringPi/WiringPi #294: wiringPi: no access to /sys/bus/pci/devices/0000:01:00.0/resource1 try /dev/gpiomem0

ZachGarcia42, 2024-11-08:
Hello! I am using WiringPi on a Pi 5 2GB running 64-bit Raspbian. PWM commands fail with the error: 
`wiringPi: no access to /sys/bus/pci/devices/0000:01:00.0/resource1 try /dev/gpiomem0`. 

That file does not exist at that path. I see resource and resource0 at /sys/bus/pci/devices/0000:01:00.0/ but no resource1.
Looking at [the source code](https://github.com/WiringPi/WiringPi/blob/master/wiringPi/wiringPi.c#L225), my dmesg gives a different result than the one in the comment: 
`[    0.664887] rp1 0001:01:00.0: bar1 len 0x400000, start 0x1f00000000, end 0x1f003fffff, flags, 0x40200`
 instead, and /sys/bus/pci/devices/0001:01:00.0/resource1 does exist on my system. I can't symlink to it without recompiling the kernel, and I haven't quite figured out how to compile from source to a 32-bit binary (which I need for unrelated reasons). 

Any fixes or workarounds would be greatly appreciated!

ZachGarcia42, 2024-11-08:
Update, I recompiled on a fresh 32-bit install with the address changed to `/sys/bus/pci/devices/0001:01:00.0/resource1`, and that worked!

mstroh76, 2024-11-09:
Hi 

Do you know what device is using 0000:01:00.0 ?
Is anything connected to PCIe connector?

I have to implement a search routine for the RP1 device PCIe adress, I thought this is always the same.

ZachGarcia42, 2024-11-09:
Hi, thanks for the response! I have a Pimoroni NVME base with a m.2 SSD that's likely using the first slot.

mstroh76, 2024-11-09:
Oh, i should have tested this, but i wave no PCIe hardware.
I will try to fix it, this weekend. Please be ready for testing.

ZachGarcia42, 2024-11-09:
Sure thing! Happy to test, and thanks for the quick response!

mstroh76, 2024-11-10:
Hi, please test the version 3.11 from branch pcie. 
It will report all pcie operations:
 
```
gpio -v

gpio version: 3.11
Copyright (c) 2012-2024 Gordon Henderson and contributors
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Hardware details:
  Type: Pi 5, Revision: 00, Memory: 4096MB, Maker: Sony 

System details:
  * Device tree present.
      Model: Raspberry Pi 5 Model B Rev 1.0
Open file: /sys/bus/pci/devices/0000:01:00.0/device
  device: 0x0001
  >> correct
Open file: /sys/bus/pci/devices/0000:01:00.0/vendor
  vendor: 0x1de4
  >> correct
RP1 device memory found at '/sys/bus/pci/devices/0000:01:00.0/resource1'
  * Supports full  user-level GPIO access via memory.
  * Supports basic user-level GPIO access via /dev/gpiomem.
  * Supports basic user-level GPIO access via /dev/gpiochip (slow).

```

Is this version working correct with your setting?

ZachGarcia42, 2024-11-10:
This version works for me; thanks for the help! Confirmed gpio command functionality and libwiringPi.so linking both function as expected. Here's the listed pcie operations:

```shell
$ gpio -v

gpio version: 3.11
Copyright (c) 2012-2024 Gordon Henderson and contributors
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Hardware details:
  Type: Pi 5, Revision: 00, Memory: 2048MB, Maker: Sony

System details:
  * Device tree present.
      Model: Raspberry Pi 5 Model B Rev 1.0
Open file: /sys/bus/pci/devices/0001:00:00.0/device
  device: 0x2712
  >> wrong
Open file: /sys/bus/pci/devices/0000:01:00.0/device
  device: 0x2263
  >> wrong
Open file: /sys/bus/pci/devices/0000:00:00.0/device
  device: 0x2712
  >> wrong
Open file: /sys/bus/pci/devices/0001:01:00.0/device
  device: 0x0001
  >> correct
Open file: /sys/bus/pci/devices/0001:01:00.0/vendor
  vendor: 0x1de4
  >> correct
RP1 device memory found at '/sys/bus/pci/devices/0001:01:00.0/resource1'
  * Supports full  user-level GPIO access via memory.
  * Supports basic user-level GPIO access via /dev/gpiomem.
  * Supports basic user-level GPIO access via /dev/gpiochip (slow).
```

closed 2025-01-05 by mstroh76