WiringPi/WiringPi #302: Support for Pi500

mstroh76, 2025-01-05:
Currently not possible due to a lack of hardware.

mstroh76, 2025-01-06:
I Pi500 branch will be prepared to make it possible for someone to test it

DeeJay, 2025-01-20:
CM5 Lite, 8G. (CM5008000) - RasPiOS 64-bit Bookworm.

```
System Information
------------------

Raspberry Pi Compute Module 5 Lite Rev 1.0
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage4

Linux cm58lbw64dt 6.6.62+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux
Revision	: d041a0
Serial		: 12b2113df46bad1c
Model		: Raspberry Pi Compute Module 5 Lite Rev 1.0
Throttled flag  : throttled=0x0
Camera          : vc_gencmd_read_response returned -1 error=1 error_msg="Command not registered"

```

WiringPi repo cloned with -
`git clone -b Pi500 --single-branch --depth=1 https://github.com/WiringPi/WiringPi.git`

Builds OK.

APT install of the .deb seems to work but throws an information message -

```
david@cm58lbw64dt:~/Support/WiringPi/WiringPi $ sudo apt install ./wiringpi_3.13_arm64.deb 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'wiringpi' instead of './wiringpi_3.13_arm64.deb'
The following NEW packages will be installed:
  wiringpi
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/68.1 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 /home/david/Support/WiringPi/WiringPi/wiringpi_3.13_arm64.deb wiringpi arm64 3.13 [68.1 kB]
Selecting previously unselected package wiringpi.
(Reading database ... 148722 files and directories currently installed.)
Preparing to unpack .../wiringpi_3.13_arm64.deb ...
Unpacking wiringpi (3.13) ...
Setting up wiringpi (3.13) ...
Processing triggers for man-db (2.11.2-2) ...
N: Download is performed unsandboxed as root as file '/home/david/Support/WiringPi/WiringPi/wiringpi_3.13_arm64.deb' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)
david@cm58lbw64dt:~/Support/WiringPi/WiringPi $ 

```

Runs from shell as expected -
```
david@cm58lbw64dt:~/Support/WiringPi/WiringPi $ gpio -v
gpio version: 3.13
Copyright (c) 2012-2025 Gordon Henderson and contributors
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Hardware details:
  Type: CM5 Lite, Revision: 00, Memory: 8192MB, Maker: Sony UK 

System details:
  * Device tree present.
      Model: Raspberry Pi Compute Module 5 Lite Rev 1.0
  * Supports full  user-level GPIO access via memory.
  * Supports basic user-level GPIO access via /dev/gpiomem.
  * Supports basic user-level GPIO access via /dev/gpiochip (slow).
```

```
david@cm58lbw64dt:~/Support/WiringPi/WiringPi $ gpio readall
 +-----+-----+---------+------+---+---CM5L---+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 |   -  | 0 |  3 || 4  |   |      | 5v      |     |     |
 |   3 |   9 |   SCL.1 |   -  | 0 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   -  | 0 |  7 || 8  | 0 |  -   | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 0 |  -   | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 |   -  | 0 | 11 || 12 | 0 |  -   | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 |   -  | 0 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 |   -  | 0 | 15 || 16 | 0 |  -   | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 0 |  -   | GPIO. 5 | 5   | 24  |
 |  10 |  12 |    MOSI |   -  | 0 | 19 || 20 |   |      | 0v      |     |     |
 |   9 |  13 |    MISO |   -  | 0 | 21 || 22 | 0 |  -   | GPIO. 6 | 6   | 25  |
 |  11 |  14 |    SCLK |   -  | 0 | 23 || 24 | 0 |  -   | CE0     | 10  | 8   |
 |     |     |      0v |      |   | 25 || 26 | 0 |  -   | CE1     | 11  | 7   |
 |   0 |  30 |   SDA.0 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.0   | 31  | 1   |
 |   5 |  21 | GPIO.21 |   -  | 0 | 29 || 30 |   |      | 0v      |     |     |
 |   6 |  22 | GPIO.22 |   -  | 0 | 31 || 32 | 0 |  -   | GPIO.26 | 26  | 12  |
 |  13 |  23 | GPIO.23 |   -  | 0 | 33 || 34 |   |      | 0v      |     |     |
 |  19 |  24 | GPIO.24 |   -  | 0 | 35 || 36 | 0 |  -   | GPIO.27 | 27  | 16  |
 |  26 |  25 | GPIO.25 |   -  | 0 | 37 || 38 | 0 |  -   | GPIO.28 | 28  | 20  |
 |     |     |      0v |      |   | 39 || 40 | 0 |  -   | GPIO.29 | 29  | 21  |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+---CM5L---+---+------+---------+-----+-----+
david@cm58lbw64dt:~/Support/WiringPi/WiringPi $ 
```

DeeJay, 2025-01-20:
CM5 Lite, 8G. (CM5008000) - RasPiOS 64-bit Bookworm.

Added later, after testing with a real connected LED -

In WiringPi/examples.

`make blink
`

Compiles and links correctly. The resulting executable is seen to blink an LED. Tested with WP pins 23/24/25.

I have never been a WiringPi user, so I don't have existing code of my own to test. Is there any other more exhaustive/demanding test suite the developers would like to have run on my test rig before I tear it down?

mstroh76, 2025-01-20:
Thank you very much for the test! 👍  It looks good and I think everything is OK. However, there is still a standardized unit test that you could perform. For this, you need to install 3 bridges at the GPIOs. 
For safety, I used a 470 Ohm or 1K Ohm resistor. The bridges should be 
GPIO26<->GPIO19
GPIO13<->GPIO12
GPIO17<->GPIO18.

To start the STD unit test:
 
```bash
cd WringPi/wiringpi/test/
make 
make test
```

Everything should be green.

DeeJay, 2025-01-21:
Not getting as far as executing the tests -

```
david@cm58lbw64dt:~/Support/WiringPi/WiringPi/wiringPi/test $ make
gcc -Wall wiringpi_test0_version.c -o wiringpi_test0_version -lwiringPi
gcc -Wall wiringpi_test1_sysfs.c -o wiringpi_test1_sysfs -lwiringPi
gcc -Wall wiringpi_test2_sysfs.c -o wiringpi_test2_sysfs -lwiringPi
gcc -Wall wiringpi_test3_device_wpi.c -o wiringpi_test3_device_wpi -lwiringPi
gcc -Wall wiringpi_test4_device_phys.c -o wiringpi_test4_device_phys -lwiringPi
gcc -Wall wiringpi_test5_default.c -o wiringpi_test5_default -lwiringPi
gcc -Wall wiringpi_test6_isr.c -o wiringpi_test6_isr -lwiringPi
gcc -Wall wiringpi_test8_pwm.c -o wiringpi_test8_pwm -lwiringPi
gcc -Wall wiringpi_test9_pwm.c -o wiringpi_test9_pwm -lwiringPi
gcc -Wall wiringpi_xotest_test1_spi.c -o wiringpi_xotest_test1_spi -lwiringPi
gcc -Wall wiringpi_i2c_test1_pcf8574.c -o wiringpi_i2c_test1_pcf8574 -lwiringPi
gcc -Wall wiringpi_piface_test1.c -o wiringpi_piface_test1 -lwiringPi -lwiringPiDev
david@cm58lbw64dt:~/Support/WiringPi/WiringPi/wiringPi/test $ make test
=== unit test: wiringpi_test0_version ===
/bin/sh: 7: time: not found

=== unit test: wiringpi_test1_sysfs ===
/bin/sh: 7: time: not found

=== unit test: wiringpi_test2_sysfs ===
/bin/sh: 7: time: not found

=== unit test: wiringpi_test3_device_wpi ===
/bin/sh: 7: time: not found

=== unit test: wiringpi_test4_device_phys ===
/bin/sh: 7: time: not found

=== unit test: wiringpi_test5_default ===
/bin/sh: 7: time: not found

=== unit test: wiringpi_test6_isr ===
/bin/sh: 7: time: not found

=== unit test: wiringpi_test8_pwm ===
/bin/sh: 5: time: not found

=== unit test: wiringpi_test9_pwm ===
/bin/sh: 5: time: not found

STD TEST FAILED

david@cm58lbw64dt:~/Support/WiringPi/WiringPi/wiringPi/test $ 

```

mstroh76, 2025-01-21:
you need to install time
''apt install time''

DeeJay, 2025-01-21:
Sorry for that! Didn't think to check the glaringly obvious: it never occurred to me that **time** was not bundled with RasPiOS!

DeeJay, 2025-01-21:
CM5 Lite, 8G. (CM5008000) - RasPiOS 64-bit Bookworm. CM5 is mounted on an official RP CM5 IO board.

WP v3.13 built from Pi500 branch.

All tests passed.

Transcript is probably uninteresting, but begins -
```
david@cm58lbw64dt:~/Support/WiringPi/WiringPi/wiringPi/test $ make test
=== unit test: wiringpi_test0_version ===
                          version major (  3==  3) -> passed
                          version minor ( 13== 13) -> passed

UNIT TEST STATE:  PASSED

0.00user 0.00system 0:00.00elapsed 90%CPU (0avgtext+0avgdata 1536maxresident)k
0inputs+0outputs (0major+65minor)pagefaults 0swaps

=== unit test: wiringpi_test1_sysfs ===
WiringPi GPIO test program 1 (using GPIO19 (output) and GPIO26 (input) via sys)
 testing digitalWrite, digitalRead and pullUpDnControl
toggle 4 times ...
set GPIO19 = 0 (readback 0), in GPIO26 = 0         -> passed
set GPIO19 = 1 (readback 1), in GPIO26 = 1         -> passed
set GPIO19 = 0 (readback 0), in GPIO26 = 0         -> passed
set GPIO19 = 1 (readback 1), in GPIO26 = 1         -> passed
set GPIO19 = 0 (readback 0), in GPIO26 = 0         -> passed
set GPIO19 = 1 (readback 1), in GPIO26 = 1         -> passed

WiringPi GPIO test program (using GPIO19 (input pull up/down) and GPIO26 (input) via sys)
in = down   set GPIO19 = 0 (readback 0), in GPIO26 = 0         -> passed
in =   up   set GPIO19 = 1 (readback 1), in GPIO26 = 1         -> passed
in = down   set GPIO19 = 0 (readback 0), in GPIO26 = 0         -> passed
in =   up   set GPIO19 = 1 (readback 1), in GPIO26 = 1         -> passed
in = down   set GPIO19 = 0 (readback 0), in GPIO26 = 0         -> passed
in =   up   set GPIO19 = 1 (readback 1), in GPIO26 = 1         -> passed
digitalWrite: no output (33)

UNIT TEST STATE:  PASSED

0.00user 0.00system 0:10.26elapsed 0%CPU (0avgtext+0avgdata 1536maxresident)k
0inputs+0outputs (0major+66minor)pagefaults 0swaps

```

and ends - after executing 
```
=== unit test: wiringpi_test9_pwm ===
WiringPi GPIO test program 9
PWM/ISR test (WiringPi 3.13)
```

by reporting -

```
set PWM@GPIO19 (output) back to input

Did 4 PWM GPIO tests with model 26

UNIT TEST STATE:  PASSED

0.00user 0.01system 3:41.84elapsed 0%CPU (0avgtext+0avgdata 4096maxresident)k
0inputs+0outputs (0major+224minor)pagefaults 0swaps

STD TEST SUCCESS

david@cm58lbw64dt:~/Support/WiringPi/WiringPi/wiringPi/test $ 
```

mstroh76, 2025-01-21:
that looks bad.

digitalWrite: no output (33)

is it the same result after a reboot?

please start the test manually like 
./wiringpi_test1_sysfs
and maybe 
WIRINGPI_DEBUG=1 ./wiringpi_test1_sysfs

mstroh76, 2025-01-21:
I wanted to apologize for the confusion earlier. The message was intentional ('Error wrong direction - only for fun'), and I initially misunderstood your post because the output was abbreviated. I thought something was wrong because it seemed too short. I didn’t notice the formatting on my phone. **Everything is fine now**. Thank you for the test, I will push the version to the next release.

DeeJay, 2025-01-21:
No problem from my PoV.

To avoid future confusion, perhaps the test code should emit an informative message indicating that the 'no output' value is the expected one?

But my contribution was to build and test on hardware you do not have direct access to, and that task is complete.

Although I have never needed to use Wiring Pi, thank you for your efforts to keep it available for those who do.

closed 2025-01-30 by mstroh76