WiringPi/WiringPi #305: Pi 5 / RP1 bad performance if read and write operations are alternating

mstroh76, 2025-01-05:
Report of bad performance compared to Pi4 if read and write function are alternated.
OK if only reading or only writing

**Need unit test case with performance index / timing**

mstroh76, 2025-01-06:
Pi 5 performance:
```
                                Model:  ( 23<> -1) -> passed
WiringPi GPIO speed test program (using GPIO 19) and toggle 192000000 times
 testing digital Wirte and pinMode
toggle  192 million times pin value ...
   192000000 toggle took 9.603 s, Time per toggle 0.050 us, Freq 19.995 MHz 
  GPIO digitalWrite max. frequency  (19.995==20.000) -> passed 

toggle  19 million times pin mode...
   19200000 toggle took 7.680 s, Time per toggle 0.400 us, Freq 2.500 MHz 
       GPIO pinMode max. frequency  (2.500==2.500) -> passed 

toggle  19 million times pin mode and digitalWrite ..
   19200000 toggle took 8.449 s, Time per toggle 0.440 us, Freq 2.273 MHz 
       GPIO pinMode max. frequency  (2.273==2.222) -> passed 

```

mstroh76, 2025-01-10:
Pi 5:
                               Model:  ( 23<> -1) -> passed
WiringPi GPIO speed test program (using GPIO 19) and toggle 192000000 times
 testing digital Wirte and pinMode
 192 million times digitalWrite ...
   192000000 took 9.601 s, Time per operation 0.025 us (toggle Freq 19.997 MHz) 
                       digitalWrite (0.025==0.025) -> passed 

 14 million times digitalRead ...
   14769230 took 9.552 s, Time per operation 0.323 us (toggle Freq 1.546 MHz) 
                        digitalRead (0.323==0.325) -> passed 

 22 million times pinMode...
   22153845 took 8.862 s, Time per operation 0.200 us (toggle Freq 2.500 MHz) 
                            pinMode (0.200==0.200) -> passed 

Toggle  22 million times pinMode and digitalWrite ..
   22153845 took 9.748 s, Time per operation 0.220 us (toggle Freq 2.273 MHz) 
           pinMode and digitalWrite (0.220==0.225) -> passed 

Toggle   4 million times digitalRead and digitalWrite ..
    4430769 took 9.864 s, Time per operation 1.113 us (toggle Freq 0.449 MHz) 
       digitalRead and digitalWrite (1.113==1.065) -> passed 
        Write <=> Read delay factor (3.180==3.200) -> passed 

Rasperry Pi 5:
  * digitalRead has very slow speed, much slower then digitalWrite, factor 13.0 (typical ~2.0)
  * Toggle read/write operation has slow speed, factor 3.20 (typical ~1.77) to single operation time

Pi4:
                                Model:  ( 17<> -1) -> passed
WiringPi GPIO speed test program (using GPIO 19) and toggle 240000000 times
 testing digital Wirte and pinMode
 240 million times digitalWrite ...
   240000000 took 9.730 s, Time per operation 0.020 us (toggle Freq 24.665 MHz) 
                       digitalWrite (0.020==0.020) -> passed 

 120 million times digitalRead ...
   120000000 took 9.144 s, Time per operation 0.038 us (toggle Freq 13.123 MHz) 
                        digitalRead (0.038==0.041) -> passed 

 30 million times pinMode...
   30000000 took 7.271 s, Time per operation 0.121 us (toggle Freq 4.126 MHz) 
                            pinMode (0.121==0.122) -> passed 

Toggle  30 million times pinMode and digitalWrite ..
   30000000 took 8.419 s, Time per operation 0.140 us (toggle Freq 3.564 MHz) 
           pinMode and digitalWrite (0.140==0.142) -> passed 

Toggle  45 million times digitalRead and digitalWrite ..
   45000000 took 9.734 s, Time per operation 0.108 us (toggle Freq 4.623 MHz) 
       digitalRead and digitalWrite (0.108==0.093) -> passed 
        Write <=> Read delay factor (1.767==1.770) -> passed

nachoym, 2025-01-11:
I found same issue this week:
1- Read operation takes around 325ns while Write operation only takes 25ns.   
2- Alternated read and write operation takes 1145 ns  --> which is much more time than the sum of both
This is unacceptable for the project I am involved in.  Is there any expectation for a possible fix an when could it be released ?
Thanks a lot to developers.

nachoym, 2025-01-13:
Maybe this document about RP1 peripherals helps:
https://datasheets.raspberrypi.com/rp1/rp1-peripherals.pdf
Page 36: Latency considerations

mstroh76, 2025-01-15:
"The PCIe link between RP1 and the host processor inserts an unavoidable amount of latency, typically 1μs at the design
link width and speed. This is of chief concern for applications that rely on rapid but timely write sequences, e.g. bit-
bashed protocols, or applications that need to respond quickly to the state change of a pin."

I don't think it is fixable. **If faster GPIO is needed, the Pi 4 is the better option**.
However, I will take a closer look at the problem.

mstroh76, 2025-01-30:
Hi @DeeJay If you’d like, could you run one more unit test on your hardware? 
Take the 'test' branch run ./reinstall , switch to wiringpi/test 'make clean; make' and run ./wiringpi_test7_bench
It will do a gpio performance test on GPIO 19/26. 
I think it will be the same as Pi 5 but would be nice to be sure.

mstroh76, 2025-08-21:
@Storynow  , please english: Do you have any new progress in this matter?

No progress. Using Pi4 is the solution so far.
But maybe the RP1 Chip can do little caching, if there is a solution it will be complicated.