WiringPi/WiringPi #315: Multiple Undeclared Errors When Running ./build debian

murphm52, 2025-02-11:
I am attempting to build the package on my Raspberry Pi 4 over Ubuntu 20.04 but keep getting this error. The furthest I got was cloning the Git to my SD card. Sorry if it is a simple fix, I am still learning how to work with Ubuntu

[image]

. When I run "./build debian", I get the error:

"error: 'GPIOHANDLE_REQUEST_BIAS_DISABLE' undeclared (first use in this function); did you mean 'GPIOHANDLE_REQUEST_OPEN_DRAIN'?"

I get the same errors for the functions GPIOHANDLE_REQUEST_BIAS_PULL_UP and GPIOHANDLE_REQUEST_BIAS_PULL_DOWN.

mstroh76, 2025-02-12:
See command '# Only if you know what you're doing!' in the build script.
I don't know if this is working; Ubuntu is still not officially supported.
Have you tried ``./build debian``?
I have successfully tested with Ubuntu 24.04. We will only support the current version of Ubuntu.

mstroh76, 2025-02-12:
The error are from missing defines of  "userspace ABI for the GPIO character devices" 
see https://github.com/openSUSE/kernel/blob/master/include/uapi/linux/gpio.h

What Kernel version do you have? 

But the solution is: 
**Please use latest WiringPi 2 Version**   https://github.com/WiringPi/WiringPi/releases/tag/2.61-1
We will not support Ubuntu 20.04 with WiringPi 3.

closed 2025-02-12 by mstroh76