WiringPi/WiringPi #322: Can't use ISR on WiringPi Pin 30

mbrandenburg88, 2025-02-19:
I try to use WiringPi Pin 30 as an Input. 
When i try to add the WiringPiISR i got the error->wiringPiISR: pin must be 0-27 (30)

So i can´t use pin 30 with an ISR?

mstroh76, 2025-02-19:
Hi, only 40-pin header GPIOs are supported, see [main page](https://github.com/WiringPi/WiringPi)  ``gpio readall``
I don't know what you mean with Pin 30.

mstroh76, 2025-02-19:
This is a 'do not use' / reserved pin, for the first I2C-Bus, I don't know if this if working, I will check. Are you sure you are on wPi mode? "pin must be 0-27" look like BCM mode.

mbrandenburg88, 2025-02-19:
Hey, 

I use wiringPiSetup(). This is the wiringPi mode, isn´t it?

mstroh76, 2025-02-19:
Workaround use BCM numbering ``wiringPiSetupGpio()``

closed 2025-05-18 by mstroh76