WiringPi/WiringPi #335: Donations

deavmi, 2025-03-16:
Hey all, recently started using your library as its under LGPL v3 and this makes it easier for
me to integrate into a proprietary project of mine. I do, however, want to show my gratitude
for this but I don't see any place where I can make a donation (once off or recurring).

If there is any way, please let me know :).

deavmi, 2025-03-16:
Can send XMR, BTC or whatever, I don't mind

deavmi, 2025-03-16:
> Can send XMR or BTC (lightning or on-chain). XMR easiest for me.

Or Liberapay, that also works. Many open source projects use it and I have a debit order that goes off every month. So that might be easiest for recurring.

chirndler, 2025-03-16:
Hey Tristan,

if you would like to donate to us, please do so via BTC bc1qtpka0cwqsm7lc8m5ujfl9zc7axz7dlesw5xm52
Thank you very much :)

closed 2025-03-16 by chirndler

deavmi, 2025-03-16:
> bc1qtpka0cwqsm7lc8m5ujfl9zc7axz7dlesw5xm52

Perfect, sent some via lightning etwork now - it should eventually hit the chain and be in your wallet.

deavmi, 2025-03-16:
> Thank you!

No problem