WiringPi/WiringPi #337: wiringPiISR with userdata?

PPokorski, 2025-03-20:
Hello. I'm using wiringPi in my C++ project and wanted to use `wiringPiISR` to register single callback for use with multiple different objects, depending on pin used, e.g. pin 0 calls callback of object 0, pin 1 calls callback of object 1, etc. The issue is I'm not sure how to do it elegantly.
Normally I'd do this in one of two ways:
1. Pass `this` object as `userdata` argument to `wiringPiISR`. Unfortunately API is not designed around it.
2. I could have my own thread with spinning `waitForInterrupt`. The documentation says, however, I shouldn't use `waitForInterrupt` directly.

Ideally I'd like to do something like this:
```
Object object;
void isr_callback(void* userdata)
{
    (Object*(userdata))->isr_callback();
}

wiringPiISR(PIN, MODE, &isr_callback, &object);
```

Is it possible to have behaviour like this? Or would this require changes to `wiringPi` to support such case?

mstroh76, 2025-04-04:
Thank you for your software feature request. We'll review it, and there's a possibility that we may implement it.

PPokorski, 2025-04-04:
> Thank you for your software feature request. We'll review it, and there's a possibility that we may implement it.

I noticed kind of similar functionality in `GPIO_V2_IOCTL` branch. What is the difference between master and this branch? Is any particularly recommended?

Also: would you be interested in my implementation of this feature? I think I could give it a try.