WiringPi/WiringPi #348: digitalWrite8 and wiringPiSPIxSetup in .h but not in .c

FroggyCorp, 2025-05-11:
Hi,

There is 2 function i tried to use : 
digitalWrite8 and wiringPiSPIxSetup which can be found in wiringPi.h & wiringPiSPI.h but for wiringPiSPIxSetup, it is not defined anywhere and digitalWrite8 is commented inside wiringPi.c
Resulting of :
RPI_SPI.cpp:(.text+0xbd): undefined reference to `wiringPiSPIxSetup'
RPI_SPI.cpp:(.text+0x3c7): undefined reference to `digitalWrite8'

I don't know if there is an hidden reason for that or if i miss something.

Thanks in advance,

mstroh76, 2025-05-11:
digitalWrite8 was already removed from WiringPi version 2. See the source code 
https://github.com/WiringPi/WiringPi/blob/V2/wiringPi/wiringPi.c

If you're encountering a wiringPiSPIxSetup error, you likely have two active versions of WiringPi installed. Try removing WiringPi and reinstalling it using the latest .deb file. 
https://github.com/WiringPi/WiringPi/releases/tag/3.14

FroggyCorp, 2025-05-11:
For digitalWrite8, is it possible to remove it from wiringPi.h ? It makes no sense to keep it.
I use lastest source wiringPi, there is no wiringPiSPIxSetup except in .h or i miss something :²

```
grep -rnw '.' -e 'wiringPiSPIxSetup'
./wiringPi/wiringPiSPI.h:41:int wiringPiSPIxSetup     (const int number, const int channel, const int speed) ;
./wiringPi/test/wiringpi_spi_test1_mcp3202.c:67:            FailAndExitWithErrno("wiringPiSPIxSetup", hSPI);
./wiringPi/test/wiringpi_xotest_test1_spi.c:81:        FailAndExitWithErrno("wiringPiSPIxSetup", hSPI);

```

Next-Door-Tech, 2025-05-16:
Issue with deprecated `digitalRead8()` and `digitalWrite8()` functions addressed in #350

Next-Door-Tech, 2025-05-27:
Fixed by merging #350 and #353

PR #350 (2025-06-05): [cleanup] Remove comments referencing deprecated `digitalRead8()` and `digitalWrite8()` functions
Cleanup commented-out references and implementations of deprecated functions `digitalRead8()` and `digitalWrite8()`.

No changes to non-commented code, other than removing `digitalRead8()` and `digitalWrite8()` prototypes from wiringPi.h, which were not commented out.

Partially fixes #348