WiringPi/WiringPi #378: wiringPiISR() returns success status on initialization failure

naokiiwakami, 2025-06-18:
I had this issue with version 3.14. I also tried version 3.16, still having the same issue.

When I call the following function twice by accident,

```
int init_mcp2515_interrupt() {
  // Enable pin down interrupt. Connect the RX0BF (11) pin on the MCP2515 chip
  // to the GPIO 1 pin (physical 28 pin) on the 40-pin Raspberry Pi.
  if (wiringPiISR(31, INT_EDGE_FALLING, mcp2515_handle_rx)) {
    fprintf(stderr, "ISR setup failed\n");
    return 1;
  }
  return 0;
}
```

the second initialization fails because the pin is occupied already. But the function `wiringPiISR()` returns 0.
I think the function should return -1 in this case as mentioned in the function doc.

The function prints this message on stderr, so the failure is detected internally.
```
wiringPi: ERROR: ioctl interruptHandlerV2: get line event of 1 (falling) returned error 'Device or resource busy' (-1)
```

I have a fix in my fork, can we discuss the fix, or shall I send a pull request?

mstroh76, 2025-06-19:
please create PR , but from develop (not master!)

mstroh76, 2025-08-26:
#396  too before I test the code?

naokiiwakami, 2025-08-27:
OK I'll work on it.

naokiiwakami, 2025-08-29:
Created PR https://github.com/WiringPi/WiringPi/pull/398

closed 2025-09-13 by naokiiwakami